TheCrypt0/yi-hack-v4 #103: Pairing has timed out, please restart the camera

fengfc82, 2019-07-18:
I have successfully connected my yi 1080p camera to WiFi but failed to pair it to my app. I have tried many times but still failed. The Version of the camera is 18CN. Is this the Version not supported?

MagnusPihl, 2019-07-24:
I have the same problem with my Yi Outdoor Camera.
Firmware works fine, it connects to WiFi with no problems. Everything's great except I can't add the camera in the Yi Home app.

saei00, 2019-07-26:
Hi I have same problem my version also is 18cn.
Error: timed out in pairing with the app

ozeraser, 2019-07-28:
It is because the CN proxy you're using isn't working or responding in time, try other proxies.

saei00, 2019-07-28:
Hello and thanks You had helpful tips for me I would appreciate if you can send a web site to a slave that resolves the proxy problem
Best regards