TheCrypt0/yi-hack-v4 #104: Difficulty in compiling

Thecodeboxed, 2019-07-20:
Hey everyone !
   I am actually trying to learn and understand how this code works and what all modules are being used . I am new to this firmware coding methods and eager to learn how this goes on .
Can U please tell what does this errors mean?
PS- I installed linux-gnueabi-gcc and soft linked it to linux-hisiv300 .
Can this be reason for the error?
[image]
[image]

shxdow, 2019-07-28:
Hi,
I'll ask you to do a couple things, such as posting screenshots (they're easier to read than photos taken with a smartphone) and try to go through the installation process using a Unix system. I'm having problems myself going through some steps using Ubuntu 18.04 LTS, so I don't expect anything on Windows 10 to work. Besides, you can give Debian 10 a try, which is what Crypto is currently using.

TheCrypt0, 2019-07-29:
Hi @Thecodeboxed,
I'm glad you are trying to understand the code and maybe help with the development!

You mentioned that you have installed linux-gnueabi-gcc and linked it to linux-hisiv300, that's probably the first reason why you are getting some errors. 

You need to download and install the proprietary SDK which comes with the cross compiler for those cameras.

Download Link: https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr/folder/18989615567

The files you are looking for are the six zip parts of: Hi3518E_V200R001C01SPC040.*

Keep in mind that the SDK is compiled for 32 bit systems, if you are using a 64 bit system you'll need to install the i386 libs. A quick google search for "linux run 32 bit on 64 bit" will help you. 

See this Wiki article too: https://github.com/TheCrypt0/yi-hack-v4/wiki/Project-Setup-Hints

Cheers