TheCrypt0/yi-hack-v4 #106: Restore original firmware as yi-hackv3

gfucka, 2019-07-24:
Hi,
To restore original firmware can I use the same firmware and procedure described in the v3 readme??  Thanks

cszuo, 2019-07-31:
And also, which firmware should we use to restore for camera 47US?

ketpa, 2020-01-27:
I have the same problem. Help.

dedelekid, 2020-09-01:
> 
> 
> And also, which firmware should we use to restore for camera 47US?

Hi, just use the method in v3 : https://github.com/shadow-1/yi-hack-v3 
Look to the table :  https://github.com/shadow-1/yi-hack-v3#going-back-to-stock-firmware
Then connect the app and let to do an upgrade.
This work for me:  ver 47us.

idanny76, 2021-05-30:
Trying to go back to stock firmware, but I cant seem to figure it out. I have downloaded the stock firmware and placed in the sd card root. But after rebooting it has the Yi Hack V4 firmware.

idanny76, 2021-05-31:
I figured it out. The stock firmware after it is downloaded it has be renamed to a specific name. PDF file here shows names of files depending on the camera version.

https://oss.yitechnology.com/images/support/homecam2usermanual.pdf

https://www2.yitechnology.com/firmware/index