TheCrypt0/yi-hack-v4 #109: Firmware not flashing, new serial number?

peely, 2019-07-29:
Hi all. I've just bought the "YI Cloud Dome Camera 1080P HD" (https://www.amazon.co.uk/gp/product/B07CTH3MBN). The serial number on the bottom starts with 8FUS, so i'm not sure its the same camera as any that are listed.

I tried putting an SD in with _h20 on, but it ignore them. I've seen [this Issue](https://github.com/TheCrypt0/yi-hack-v4/issues/49), but i'm not sure its for the same camera. Am I downloading the wrong firmware or have I got a new variant of this model of camera?

Any help appreciated, i'm just trying to get access to the cams video stream, so I can save it to my server instead of paying for the cloud ha. Thanks.

DJBenson, 2019-07-29:
Subscribing to this thread on the basis I just ordered 2 of these cameras (same ASIN as above) from Amazon to hack.

DJBenson, 2019-08-01:
Mine arrived from Amazon today with the 8FUS...serial, not tried flashing it yet but not overly hopeful.

DJBenson, 2019-08-01:
As expected, the firmware is not read from the SD Card.

For reference, this is the model we are discussing, it appears to be a revised version based on the case.

[image]

jamieshaw, 2019-08-04:
I purchased this camera also (during the 33% reduction offer); same serial number of 8FUS; and the hack works like a charm – got it running with ZoneMinder and that feeding into Home Assistant without issue.

It's worth noting that the square bottomed model pictured above is the Yi _cloud_ dome, not the Yi dome – it requires **h19** opposed to h20.

peely, 2019-08-05:
"Yi **_cloud_** dome"

I honestly hope I've been this stu... situationally unaware. Will try tonight.

peely, 2019-08-05:
Yes, @jamieshaw you were right. I just didn't read 😞 Thank you for being nice in pointing out my mistake.

closed 2019-08-05 by peely

jamieshaw, 2019-08-05:
No problem – these things happen, glad I could help 😄