TheCrypt0/yi-hack-v4 #113: Unable to pair to MI Home app

ivanonghm, 2019-08-04:
Camera: 17CN

Pardon the question and based on what I've been reading on the WIKI, am I right to say that unless I find a work PROXY for use with proxychain-ng, the camera technically will no longer work with the app even though it's paired with the app?

Crescendo-BLYAT, 2019-08-04:
https://www.xroxy.com/free-proxy-lists/?port=&type=All_socks&ssl=nossl&country=CN&latency=&reliability=

works for me....

ivanonghm, 2019-08-05:
Thanks for the share.

May I know what else do you have to do to make the app work as the app still show that the camera is offline, please reconnect it (-3006).

Appreciate if you can provide some directions.

Thank you.

ivanonghm, 2019-08-05:
I've fixed it since going through the instructions from:
[](https://github.com/Rohja/yi-hack-v4/commit/08b984cb01402a5f62e5d74b187c4ac414bc70b5)

As well as using the proxy list provided by you.

Thank you.

ivanonghm, 2019-08-05:
One thing I noted was that after disconnecting the app.
Using 4G to connect over the internet, I can no longer connect to the camera via the app or even so after I connect back to the internet at home...