TheCrypt0/yi-hack-v4 #114: Any news on ONVIF support?

itm1960, 2019-08-06:
I've just discovered this, and I'm using it with a Yi Cloud Dome 1080P - great work!
I was wondering if there was any news on ONVIF support? It would be sooooo useful to use PTZ without having to use the manufacturer's app.

justin-vt, 2019-08-13:
+1

This would be the killer app, creating an inexpensive 1080p HD PTZ camera with the capabilities of the Foscam / Amcrest / <insert your generic 2008 PTZ camera here> firmwares.

May relate to https://github.com/TheCrypt0/yi-hack-v4/issues/116#issue-479400218

jamieshaw, 2019-08-20:
Just tacking myself on here to keep up with any updates; basically looking for PTZ support on the Cloud Dome – not entirely sure if this is something currently supporting via command_line, HTTP, or otherwise. Have the features been unearthed within the firmware at all?

bdabda55, 2019-10-05:
Hi just adding my vote for PTZ support (on the Yi Dome, but I plan to acquire a Yi Dome 1028 as well, once PTZ is available on this GREAT hack). 
Thanks.

fidesachates, 2019-12-18:
Adding request for PTZ support

stevepesch, 2020-01-16:
Great hack! Adding PTZ will be the icing on the cake.

grungeguitarist, 2020-01-17:
Yes, that would be awesome (besides fixing the unstable RTSP and high CPU usage).

ajdezigns, 2020-03-08:
Has this project stalled out?

Zippeyrude, 2020-04-05:
Fantastic effort
Any news on ONVIF please?

Zippeyrude, 2020-04-05:
... I saw that https://github.com/roleoroleo/yi-hack-MStar has the support but for a different serial number, is this functionality difficult / possible to port across? :)

Keroi, 2020-05-03:
Hey guys, I recompiled the firmware with PTZ support from WebUI and some extra settings (led, ir, ...) based on the work by roleoroleo (https://github.com/roleoroleo/yi-hack-MStar). Tested on 2 Yi Cloud Dome 1080p and it's working great!  Not tested on other devices but it should works...

If you're interested, here's the release : https://github.com/Keroi/yi-hack-v4/releases

pkishino, 2020-05-03:
@Keroi I'll try your release..just put files on SD and run, right?
this repo is pretty much dead..at this point is is quite ridiculous to keep charging money when no further work is done

pkishino, 2020-05-04:
> Hey guys, I recompiled the firmware with PTZ support from WebUI and some extra settings (led, ir, ...) based on the work by roleoroleo (https://github.com/roleoroleo/yi-hack-MStar). Tested on 2 Yi Cloud Dome 1080p and it's working great! Not tested on other devices but it should works...
> 
> I also tried to add onvif support but with no luck :(
> 
> If you're interested, here's the release : https://github.com/Keroi/yi-hack-v4/releases

Works well,pzt is useful. 
Camera on/off as well.. have to see if I can trigger this remotely. 
Status led is always off now, the setting doesn’t turn on, only issue I found so far

accelle17, 2020-05-04:
I had to re upload the license file and use port 554 directly on the RTSP URL now. PTZ is working but not the bookmark/presets (I have 4 bookmarks on yi app). Haven't tested if recording with motion will work when cloud is disconnected.

Keroi, 2020-05-04:
> Works well,pzt is useful.
> Camera on/off as well.. have to see if I can trigger this remotely.
> Status led is always off now, the setting doesn’t turn on, only issue I found so far

You can try the command `ipc_cmd --led ON` to turn on the LED on the camera. I suppose, depend on the model, the command may not work.

pkishino, 2020-05-04:
Command not found.. is there a way we can raise issues on your fork instead of here? The project here is pretty much dead so should move over discussions etc

Keroi, 2020-05-05:
> Command not found.. is there a way we can raise issues on your fork instead of here? The project here is pretty much dead so should move over discussions etc

I'm not planning to continue development, PTZ was kind of the only thing that I was missing. It was easy to add this feature as the work had been already done by roleoroleo.
But if I can help or if there's something wrong in the release, feel free to create an issue, I'll see what I can do!

guillaume-smo, 2020-05-05:
@Keroi , would it be possible to do a pull request to TheCrypt0 git to easily keep track of your devs if you intend to stop developing your fork please ? thanks !

Keroi, 2020-05-05:
> @Keroi , would it be possible to do a pull request to TheCrypt0 git to easily keep track of your devs if you intend to stop developing your fork please ? thanks !

You're right. Done

accelle17, 2020-05-05:
looks like the preset is working via command line "ipc_cmd -p #". On other note, recording is not working when:

Disable Cloud = on  
Recording without Cloud = on
Save video when a motion is detected = on

However, I've experience this on original 0.2.0. Just wanted to check if someone was able to record when there's motion and cloud disabled.

Keroi, 2020-05-05:
> looks like the preset is working via command line "ipc_cmd -p #"

OK. There's a `preset.sh` file missing (this file : https://github.com/roleoroleo/yi-hack-MStar/blob/master/src/www/httpd/cgi-bin/preset.sh). I will publish a new release fixing this issue later today.

roleoroleo, 2020-05-07:
> I also tried to add onvif support but with no luck :(

What's the problem with onvif?

odelma, 2020-11-25:
Am I interpreting correctly that onvif is implemented in this repo: https://github.com/chautruongthinh/yi-hack-v4 (https://github.com/chautruongthinh/yi-hack-v4/tree/master/src/wsdiscovery/wsdiscovery/onvif). But there are no releases available. It would be great if someone with some knowledge about these could combine the features added by Keroi and from this (and maybe other) repos? It seems that Crypt0 is not actively doing any development anymore.

K2M7L, 2023-05-17:
up, any update on onvif?