TheCrypt0/yi-hack-v4 #118: Can't connect to web interface

hinchy34, 2019-08-16:
I have the Yi 1080p Home Camera, and using a 64gb microsd (not sure if it's too big).

It's formated to FAT32 (not exFAT), the allocation unit size was 4kb.

I grabbed yi_hack_v4_v0.2.0__yi_dome_1080p.zip and I confirmed the two files are in the root directory of my microSD card. I did not include the folder they came in yi_home_1080p.

The camera is off, put in the card and power it up. It does the yellow light then blue comes up.
I am able to use the mobile app to connect to the device, I can also check the network settings to find the local ip of 192.168.0.24 however I am not able to connect to the web interface at this step.

I am able to ping 192.168.0.24 sucessfully. So I am not sure where I went wrong?

bepece1, 2019-08-20:
I have same problem with firmware 4.60.0.0A

[https://github.com/TheCrypt0/yi-hack-v4/issues/119](url)

bepece1, 2019-08-20:
@hinchy34 I think the problem is the filename...
Review my comment
https://github.com/TheCrypt0/yi-hack-v4/issues/119#issuecomment-522973343

shxdow, 2019-08-28:
@bepece1, @hinchy34  did changing the filenames solve your issue ? Also, can you guys run a port scan on the camera to at least make sure port 80 is open ?