TheCrypt0/yi-hack-v4 #119: yi-Hack-v4 not success in 4.60.0.0A

bepece1, 2019-08-20:
I can't hack my Yi Cam (1080 dome) with firmware (4.6.0.0A_201903141924)
I copied the both files in root sd card and boot cam but yi-hack-v4 not install so as the original firmware continues in cam.

Please help

bepece1, 2019-08-20:
@TheCrypt0 
The firmware filenames for the Yi 1080p Dome camera must be home_h20 and rootfs_h20?? sure??

I'm reading in internet forum... I think the right form is "home_h20m" / "rootfs_h20m"

Can you confirm?? and fix....

Thankssssss

ykhandler, 2019-08-23:
Please try to reset it, put sdcard into the camera, and pair it again to the app, wait until it success, then turn it off, and send the log file that appear in the sdcard...

bepece1, 2019-08-26:
no work...

please, tell me something.

thanks

raystlin, 2019-08-26:
Please, read the docs...

https://github.com/TheCrypt0/yi-hack-v4/wiki/Supported-Camera-Models

shxdow, 2019-08-28:
As @raystlin has suggested, Yi 1080p Dome 6FUS is currently work in progress. I can't tell you exactly which features you could get working as of right now. Flashing should work as intended though

bepece1, 2019-08-31:
"6FUS" is "Yi Dome 1080p" or "Yi Home 1080p"???

bepece1, 2019-09-01:
> "6FUS" is "Yi Dome 1080p" or "Yi Home 1080p"???

[image]
 Yi Dome 1080 has just published a new oficial firmware

nikifena, 2019-09-09:
The same problem 
I can't flash my 6FUS camera.

The yellow led does not blink at startup. 
I use 16Gb FAT32 formatted card and both home_h20 and rootfs_h20 files were copied directly on root directory on the card.

Any ideas?

bepece1, 2019-09-23:
no updates in DOME 1080p [model 6FUS] ????

ozzeruk, 2019-10-03:
I have the same issue - 4.6.0.0A on Yi Dome 1080p won't accept the custom firmware. Does anyone have a workaround? Maybe forcing it back to an earlier version of the official firmware or something. I was happy to donate and contribute - but I want to see the custom firmware on my device first.

GuitarSkater, 2019-10-14:
same issue..... ive tried multiple firmware versions to no avail

bepece1, 2019-11-04:
@roleoroleo just published a folk 

[https://github.com/roleoroleo/yi-hack-6FUS_4.5.0](url)

TDPFlorida, 2019-11-06:
I bought a couple of "YI 1080P Dome Cameras (USA)" a couple of years ago and installed the yi-hack-v3 0.1.5 and everything worked great. I just bought a new "YI 1080P Dome Cameras (USA)" and the firmware is "4.6.0.0A" and am unable to install the same hack. I thought maybe it was because the firmware was so high (the other 2 are at "1.9.2.0l" which appears the latest) but cannot change the firmware on the new one at all. How do you determine if it is:
6FUS / 9FUS / BFUS

Thanks

TDPFlorida, 2019-11-16:
See above...how do I determine if the YI 1080 Dome (USA) is a 45US, 65US, 6FUS, MSTAR SoC, Version 1 or 2?
Thanks

ikjl, 2020-06-09:
I'm having the same issue.
I YI Dome Camera 1080 boots as normal. The yellow light is solid and odes not flash, then it goes to blue and the came seems to remain in default operation.

GuitarSkater, 2020-06-09:
> I'm having the same issue.
> A YI Dome Camera 1080 boots as normal. The yellow light is solid and does not flash, then it goes to blue and the camera seems to remain in default operation.
> Firmware version is 4.6.0.0A_201908271549

What version of the camera do you have? On the bottom it will say it. BFUS, 6FUS, etc

feeling001, 2020-06-25:
Same behavior here, unable to flash home_h20 and rootfs_h20 on YHS.2020 model (yi dome 1080)
The camera start normally. (without any orange blinking led).
(I tried the yi_dome_1080p firmware).

GuitarSkater, 2020-06-25:
Use this version.

https://github.com/roleoroleo/yi-hack-MStar

feeling001, 2020-06-25:
> 
> 
> Use this version.
> 
> https://github.com/roleoroleo/yi-hack-MStar

Thanks , it works like a charm ;-) 

Regards !

xxmasoxx, 2021-01-14:
guys i have problem with dome 720p 
model is 63us  firmware 1.9.1.0J i tried many firmw but i cant able to activate the hack
Who can help me?

idanny76, 2021-06-07:
> Who can help me?

Did you ever figured it out? What firmware were you able to flash?

veverkap, 2021-07-05:
I can't get it running either.  YHS.2016 and 15US

[image]

idanny76, 2021-07-06:
> I can't get it running either. YHS.2016 and 15US
> 
> [image]

Try using Yi-Hack V5. That's what I used for all my Yi cameras.  There is also MStar version. Check what firmware the camera is using.

https://github.com/alienatedsec/yi-hack-v5

https://github.com/roleoroleo/yi-hack-MStar