TheCrypt0/yi-hack-v4 #120: Configure RTSP + Audio on Synology SurveillanceStation

slavikme, 2019-08-20:
Hi,

I am trying to add a camera to Synology Surveillance Station using RTSP protocol, but never been able to add audio. Despite the fact that Surveillance Station does support AAC format.
I think the problem is that the audio stream is in a separate path, which is impossible to configure on most of the RTSP clients, like Surveillance Station.

Any help would be appreciated.
Thanks!

bepece1, 2019-08-21:
> Hi,
>
> I am trying to add a camera to Synology Surveillance Station using RTSP protocol, but never been able to add audio. Despite the fact that Surveillance Station does support AAC format.
> I think [...]

What camera model?

slavikme, 2019-09-11:
Yi Dome Camera 1080p

coalfield, 2020-02-01:
same issue here, same camera.  Is it possible to add audio to surveillance station?

fregol, 2020-04-11:
I’m in the same situation. Same camera model. 
I was ai to configure it on Surveillance Station, it’s working fine, but no audio. 
Any news or update?

coalfield, 2020-04-11:
kind of frustrating as its essentially useless without it for me

fregol, 2020-04-11:
> kind of frustrating as its essentially useless without it for me

Just wonder if there is any other custom firmware that allow to use this cam in surveillance station with audio (and pan why not!)

Vannixxo, 2020-05-17:
Yes please, I would also need the audio to be embedded into the video h264 and not a separate stream. I've got another camera with Mstar YI hack and it has audio embedded and works flawlessly.

thedeany, 2020-10-27:
I'd also really like to see audio embedded into the same RTSP stream as the video. I've been trying workarounds to merge the two streams into one via a proxy, but it's not reliable. Having the audio in the stream right from the source is the best case scenario. Perhaps it could be an option in the RTSP configuration so those who want the video-only stream can still get that while those of us who'd like video and audio in a single stream can toggle it on.

As mentioned above, the [MStar version has audio added to the stream](https://github.com/roleoroleo/yi-hack-MStar/pull/171).