TheCrypt0/yi-hack-v4 #126: not able to flash Yi Home 1080p

littlesteve01, 2019-08-22:
Hello,
you tried to flash my brand new Yi Home 1080p which i bought on Amazon in Germany but it isn't working... :( 
current Firmware which is shown in the YiHomeApp is: 4.5.0.0A_201902261502

can you tell me what´s wrong in my doing or is it not working for that Firmware?

marcotuna, 2019-08-24:
I am having exactly the same problem. I do have the same camera and it is using the same firmware version.

marcotuna, 2019-08-26:
Hi @Christian-Krupa, this firmware still does not support the latest version of the Yi Home 1080p
You can take a look on this comment showing that there is still work in progress, it has some instructions if you would like flash a Yi Home 1080p and start contributing.
https://github.com/TheCrypt0/yi-hack-v4/issues/128#issuecomment-524630624

Christian-Krupa, 2019-08-26:
will have a look. thx @marcotuna

untotren, 2019-09-02:
Same issue here

nitinjs, 2019-09-22:
Does not work on my YI 1080p dome...

steef84, 2019-09-27:
there is a alpha release here https://github.com/roleoroleo/yi-hack-6FUS_4.5.0