TheCrypt0/yi-hack-v4 #128: RTSP for mstar platform

roleoroleo, 2019-08-24:
I correctly compiled all sources for the 6FUS camera.
Now I would like to enable RTSP server.
Since it is closed source, is there a binary compiled for mstar platform?

marcotuna, 2019-08-24:
Hello @roleoroleo, can you provide the files to flash the camera (6FUS)? Or explain the steps you took? I have this camera and I would like to contribute
Thanks!

roleoroleo, 2019-08-25:
I don't have a final solution yet, but if you want to try I can give you a rootfs file system with a telnet access: https://repository.ilpuntotecnico.com/files/roleo/public/yi_home_1080p/sys_y203c.zip
My version is 4.5.0.0A_201902261502
Unzip and copy it to the SD card, the name of the file must be sys_y203c. U-boot will copy the content in flash.
I have also a home partition with dropbear, ftpd etc... but it's work in progress.

The steps I followed:
1 - I have unsoldered the qh128a flash and dumped it
2 - I got the various partitions and unpacked them
4 - I installed the SDK https://drive.google.com/file/d/1DnTe8G4FV4rMuopYFrdX6sx_ZeFv9YMD/view?usp=sharing
5 - I downloaded the git yi-hack-v4 and modified everything I needed to compile correctly
6 - I repacked the jff2 file system and loaded via SD

steef84, 2019-09-08:
Very nice, do you have an eta for the next version?

roleoroleo, 2019-09-19:
I completed a first beta release:
https://github.com/roleoroleo/yi-hack-6FUS_4.5.0
USE AT YOUR OWN RISK.

15goudreau, 2019-09-19:
@roleoroleo Really great work you're doing! My main purpose is getting RTSP which seems like it is locked behind closed doors. So it doesn't seem like that is going to change any time soon correct? If that is the case, I'm going to look into other cameras like Wyze to get cheap RTSP.

roleoroleo, 2019-09-19:
In this first beta, RTSP is working but not perfectly.
Sometimes a frame is wrong and then the video breaks until the next i-frame.

The problem is that I'm not a video encoding expert.
The frames of the high resolution video and those of the low resolution video are mixed.
I can distinguish i-frames thanks to the SPS, but not p-frames.
Assuming that the hi-res frame video comes before the low-res one I manage to extract them by looking at the poc field. But this procedure is not error free.

If someone wants to help me the code is in the ByteStreamFileSource.cpp file.

SurfingSpider, 2019-10-07:
I know there is a fork that has it working with the 6fus cameras, but do you plan on integrating this into the yi-hack-v4? I really like this project and how it is being actively developed and would like to stick with it on all my cameras

roleoroleo, 2019-10-08:
I don't know.
I created a new project because I think it is very complicated to prepare a tree that compiles for two different hardware platforms.
There are many different things: uClibc, uClibc++, RTSP...

skylarhays, 2019-10-21:
Why isnt this working on 6fus 4.6 firmware can you please get this working for that?