TheCrypt0/yi-hack-v4 #129: RTSP - How do I conntect from outside my network?

Kilberz, 2019-08-29:
So I've setup port forwarding on 80 and I can see the yi-hack-v4 status page from outside my network and locally but when I try to connect via rtsp outside my network I can't get the feed, but rtsp is working locally.

DoraDeu, 2019-08-29:
For the RTSP stream itself there is the need to forward one or two port more.

Kilberz, 2019-08-29:
Does anyone know what they are, please?

DoraDeu, 2019-08-29:
Sorry, I don´t. But forwarding 554 TCP and 6970 -6999 UDP could be a good starting point.

nicmorais, 2019-09-16:
The camera streams at port 554, the default port for RTSP streams. However, you can forward any port you want to in your router.
For example, you can forward the port 558 to 554:
WAN > port 558 in the router > port 554 in the camera