TheCrypt0/yi-hack-v4 #135: Stream unavailable after some time

jokerigno, 2019-09-05:
Hi,

I flashed correctly 2 outdoor cam last month and I have an issue on one of them.

After some time one stream become unavailable (only for 1 cam, the other works fine).

I check web page and everything seems ok. Only solution is to reboot camera et voilà I can see stream again.

Please let me know which information do you need to investigate.

Thank you in advance

MrTJay, 2019-09-06:
This happens me too. With a new Black dome 1080k.

The camera stoppes streaming and just a reset will fix it.

jokerigno, 2019-09-06:
I don't use yi app so I think it's not relelated (just to start thinking on the bug)

monkeydust, 2019-09-16:
Also have this issue as well - really annoying for what otherwise is a great hack. Could the developer investigate?

jokerigno, 2019-09-16:
I think he's not so interested on mqtt sensor.

michaelkleinhenz, 2019-10-18:
Check your power supply. I had issues similar to this and the reason was that the power supply wasn't delivering enough amps (or something else, the power supply caused the cam to crash). As you only have the issue with one cam and it is appearing over time, I would definitely try to swap the power supplies and see if the issue now appears on the other camera.

jokerigno, 2019-10-18:
I found out that in my case was a connection issue. I had to put a poweline transmitter next to the webcam to have video streaming restore.

The fun thing is that webpage still works and there is no error on this.

jokerigno, 2019-10-23:
self edit: maybe it would be helpful to have visibiity of signal strenght on status page. Or it would be better to add wifi scan functionality to choose best AP in the same SSID