TheCrypt0/yi-hack-v4 #136: Slow FTP speeds

dj-matt-e, 2019-09-06:
Thanks for the hack. It installed exactly as per the instructions for me. Now I am trying to download around 1.5 hrs worth of footage for a neighbour via FTP. It starts transferring quick at around 500kbps per second for the first 2 or 3 files but then suddenly slows to 27kbps for everything thereafter.

Any ideas what is causing this?

nicmorais, 2019-09-16:
Check the load of his wireless network (how many clients are associated, bandwidth) and the signal strength.

dj-matt-e, 2020-09-08:
Today I found the solution to my problem. It wasn't the wireless network at all causing the bottleneck.

The USB plug I had powering it was an old iPhone 5V 1A supply. I swapped it today for a 5V 2.4A supply and suddenly my issues have gone. It seems like the 1A supply was under powering the camera the whole time. With the 2.4A adapter everything is working great.