TheCrypt0/yi-hack-v4 #141: Yi Home (720p/USA) Dropping Frames

NoNaym, 2019-09-17:
While monitoring my old Yi Home camera via various devices, the camera appears to be dropping about 30 seconds worth of frames every 60-90 seconds.

Not sure if this is a wifi related issue or just a hw/sw issue... Is there any way to check the wireless signal? Would be awesome if this could be shown in the Web GUI somehow.

Any other thoughts though? Really loving this SW so far @TheCrypt0, very promising!

andrea1186, 2019-10-01:
@ TheCrypt0 Ciao Davide possiamo sentirci in privato?

NoNaym, 2020-01-25:
Originally closed this issue because I was unsure if this was just a local network problem... but I am definitely still seeing the same issue after various diagnostics and fixes. I think this issue might be related to #173

NoNaym, 2020-02-17:
Appears to already be discussion on this issue in #89

closed 2020-02-17 by NoNaym

idanny76, 2022-01-23:
I was able to get the problem solved, thanks to RoleoRoleo from yi-hack Mstar. He was able to update or modify the rtsp server and it now automatically connects immediately when something may go wrong with the live feed. Below is the link of the long chat troubleshooting the issue and how it was solved.

roleoroleo/yi-hack-MStar#414