TheCrypt0/yi-hack-v4 #148: does this guide work on yi-hack-v4

michael11123, 2019-09-30:
https://github.com/bobi/xiaomi_gdrive
please support me.
thankyou alot.

michael11123, 2019-10-01:
@TheCrypt0  please help me.

avengermsoft, 2019-10-11:
For the scripts to work, you need to add a <wc, stat, egrep, expr, basename, sort> to the BussyBox

avengermsoft, 2019-10-11:
Ask @TheCrypt0 add modules to the next firmware release

avengermsoft, 2019-10-13:
Add modules in pull request https://github.com/TheCrypt0/yi-hack-v4/pull/62

michael11123, 2019-10-21:
> modu

so, i cant't use this guide yet?

avengermsoft, 2019-10-22:
Try this mod [xiaomi_gdrive](https://github.com/avengermsoft/xiaomi_gdrive)

michael11123, 2019-10-23:
> Try this mod [xiaomi_gdrive](https://github.com/avengermsoft/xiaomi_gdrive)

special thank you. 
and i will close this issue.

michael11123, 2019-10-25:
[image]
@avengermsoft 
so sad. 
I have successfully done it the first time.
but the second time it reported an error.

michael11123, 2019-10-26:
so, i used FileZilla to put  "gdrive" to home/hd1 ( i creat new file hd1).
and then:
[image]

michael11123, 2019-10-26:
i creat file /home/libusr and then
[image]

michael11123, 2019-10-26:
> Try this mod [xiaomi_gdrive](https://github.com/avengermsoft/xiaomi_gdrive)

can this mod work on yi 1080 or dome?

avengermsoft, 2019-10-26:
Where did you get the curl? My curl does not require libz.so.1. Tell me the size of the curl - 
`ls -al /home/hd1/gdrive/yi-hack-v4/bin`

michael11123, 2019-10-26:
I have reinstalled by your mod and worked.

michael11123, 2019-10-27:
i have the problem. time on record folder is not same on device.
i use GMT + 7.

michael11123, 2019-10-27:
> Where did you get the curl? My curl does not require libz.so.1. Tell me the size of the curl -
> `ls -al /home/hd1/gdrive/yi-hack-v4/bin`
[curl.zip](https://github.com/TheCrypt0/yi-hack-v4/files/3775399/curl.zip)
curl file of porunov

closed 2020-03-16 by michael11123