TheCrypt0/yi-hack-v4 #149: Access to camhash

feiticeir0, 2019-10-01:
I can't access to camhash for the donation - don't know how to view it in the web page of the cam... Any help on how to access the camhash would be appreciated

npetrovski, 2019-10-02:
No 8080 port - you should access your camera's web interface via the standard port 80, a.k.a http://[your-camera-ip-address]. If you don't know your camera IP address - check you router statistics and you should be able to see it.

Another option is to try via SSH (if the SSH server is enabled in your camera), you can use PuttY or any other SSH client.
1) SSH to your camera usng username `root` with no password
2) execute `/home/yi-hack-v4/bin/camhash` and you will get the camera key

abarbra1978, 2020-01-04:
I am having the same issue. When I navigate to 192.168.1.3(as an example) the web page does not show up. When I include port 8080(192.168.1.3:8080) I get a web interface but there is no RSTP header.  

[image]

abarbra1978, 2020-01-04:
Thanks BTW!

abarbra1978, 2020-01-04:
Also @npetrovski  When I SSH into the device and navigate to /home/yi-hack-v4/bin  there is no directory or file called "camhash".  I am not linux savvy but I can get around in it(sorta).  Thanks again.

[image]

ndonna04, 2020-02-13:
@abarbra1978 mee too

Gogitt, 2020-12-17:
Well, I cannot find my camhash too, I got massage "not found" after typing /home/yi-hack-v4/bin/camhash
[image]