TheCrypt0/yi-hack-v4 #152: Status of 6fus 4.6.0 cameras?

skylarhays, 2019-10-10:
stuck with 3 of these..... really considering getting rid of them if you guys are giving  up on them. You should really make it more clear that the 6fus cams arent supported since they have the same model name as the supported cameras... i dont think you can even buy hackable yi cams anymore.

GuitarSkater, 2019-10-14:
BUMP

I am in the same boat bro

bepece1, 2019-10-14:
I think this project is abandonated :(

skylarhays, 2019-10-17:
awww..... come on....

GuitarSkater, 2019-10-17:
Seriously?! Ughhhhh, well I guess I'll be returning these new version cameras

skylarhays, 2019-10-18:
i remember back when people followed through with promises.... kids today will never understand.

dnorman12345, 2019-11-01:
I found another page that is working with this model (It is not mine, but it works on my camera):
https://github.com/roleoroleo/yi-hack-6FUS_4.5.0

GuitarSkater, 2019-11-01:
> I found another page that is working with this model (It is not mine, but it works on my camera):
> https://github.com/roleoroleo/yi-hack-6FUS_4.5.0

YOU sir, are a God amongst men. I will try it out now

GuitarSkater, 2019-11-06:
Can Confirm, https://github.com/roleoroleo/yi-hack-6FUS_4.5.0 works fantastically on 6FUS, 9FUS, BFUS cameras

bepece1, 2019-11-07:
> Can Confirm, https://github.com/roleoroleo/yi-hack-6FUS_4.5.0 works fantastically on 6FUS, 9FUS, BFUS cameras

What camera model ?

gitxplr, 2019-11-07:
Will 6FUS be added to yi-hack on the future or just use this form https://github.com/roleoroleo/yi-hack-6FUS_4.5.0 for 6FUS cameras

GuitarSkater, 2019-11-07:
> Will 6FUS be added to yi-hack on the future or just use this form https://github.com/roleoroleo/yi-hack-6FUS_4.5.0 for 6FUS cameras?

I honestly don't know why you would be requesting support for something when you have been given a solution.... And roleoroleo's fork includes rtsp and other features without a "supporter donation" 

(nothing personal TheCrypt0, you have made amazing stuff)

skylarhays, 2019-11-07:
the only problem is Roleo's hack is missing ptz controls as the cam he
initially did the hack on did not have PTZ. He assisted me with my dump but
the hack does not have PTZ. So we need someone to contribute to Roleo's
hack that can add the PTZ portion so we can move the camera's when not in
cloud mode.

benjimeistro, 2019-11-07:
@skylarhays On the readme right at the bottom Crypt0 mentions about adding ptz controls to the dome. [here](https://github.com/TheCrypt0/yi-hack-v4/blob/master/README.md)  I have a dome with yihackv4 on and I had to downgrade the firmware first, before it would flash cryptos yihackv4. It works well and I will be keeping this fw on it as crypto had already mentioned adding PTZ controls previously. 😎