TheCrypt0/yi-hack-v4 #153: 9FUS status

Hunty82, 2019-10-14:
Hi, I've recently bought a pair of Yi 1080p 9FUS camera.
There is any hope to hack them?

B1G1, 2019-10-23:
I have the same camera, i think i will return them

steef84, 2019-10-26:
https://github.com/roleoroleo/yi-hack-6FUS_4.5.0 this fork works for me on both 6fus en 9fus 1080p home camera's

roleoroleo, 2019-10-26:
Really?
9FUS also?

B1G1, 2019-10-27:
It doesn't work on my 9FUS, i lost connection to the camera even if the blue light is on.
Then if I reset it i can't pair it to the app anymore... and I don't know how to get it back to work

skylarhays, 2019-11-07:
It could be made to work if you are capable of dumping the firmware via serial. You have to solder a few connections but its not too bad.

Hunty82, 2019-11-08:
It will be not a problem. Do you have a guide?

idbartosz, 2020-03-02:
Any update on this one? I didn't test if my FUS9 works on 6FUS but I'm willing to do a firmware dump - just point me to some docs 📖

CC @skylarhays