TheCrypt0/yi-hack-v4 #154: YI Dome Camera 1080p

V1ct0rV, 2019-10-17:
Hi,

I have a YI Dome Camera 1080p model YHS.2016, I follow all the steps but nothing happens, this specific model is not supported?

baglio, 2019-10-25:
Same here:

- Formatted the microSD card in FAT32
- Placed rootfs_h20 and home_h20 in the root of the microSD
- Unplugged the camera from power and inserted the microSD
- Plugged the power to the camera

The camera just boots up as it normally does and nothing changes.

Dude88, 2019-10-29:
Hi, same here, any details?

GilDev, 2019-10-29:
Same here. I’ve got the following model though, which is advertised as Yi Dome 1080p on AliExpress and works fine with the Yi IoT app (firmware version 7.0.00.73a_201812031453) but looks different.

[image]

VitalyFedyunin, 2019-10-30:
Mine is from the amazon https://www.amazon.com/gp/product/B01CW4BLG8 same symptoms.

Dude88, 2019-10-30:
Found my problem, my serial number starts with BFUS, not 6FUS ..... Seems that there is a new revision..... , Just ordered all cameras from the local stores, hopefully there will be one with 6FUS

baglio, 2019-10-30:
> Found my problem, my serial number starts with BFUS, not 6FUS ..... Seems that there is a new revision..... , Just ordered all cameras from the local stores, hopefully there will be one with 6FUS

Interesting. How can I check my serial number? Is it located directly on the product or is it shown somewhere inside the phone app?

VitalyFedyunin, 2019-10-30:
> one on the ground of the camera

yes

GilDev, 2019-10-30:
The QR Code on mine (weird variant I posted just above) is `A0000000091WBN181011`… I think I’ll never be able to install this firmware on it, too bad… 😕

michele8gr, 2019-11-02:
> The QR Code on mine (weird variant I posted just above) is `A0000000091WBN181011`… I think I’ll never be able to install this firmware on it, too bad… 😕

Hi Gil, do you have any update ? My cam have the same label. thks

GilDev, 2019-11-02:
Well no, I don’t know if the firmware is easily adaptable and if someone will ever try to do it.
I personally don’t have time nor lots of knowledge in this area to dig more right now…

michele8gr, 2019-11-02:
> Well no, I don’t know if the firmware is easily adaptable and if someone will ever try to do it.
> I personally don’t have time nor lots of knowledge in this area to dig more right now…

I hope that someone solve the issue.
This is the printscreen when i try to access using telnet (other port are closed)

[image]

GilDev, 2019-11-02:
Nice, at least it's something!

michele8gr, 2019-11-21:
guys help us please to solve the issue. Thank you in advance.

Hataru80, 2019-12-06:
same problem... any solution?

Hataru80, 2019-12-07:
The solution for BFUSY cam is use this firmware https://github.com/roleoroleo/yi-hack-MStar

GilDev, 2019-12-09:
Damn, mine has “XY-R9820-K2(N)” on the back sticker, is there a way?

GillesC, 2019-12-09:
> Damn, mine has “XY-R9820-K2(N)” on the back sticker, is there a way?

I have the same Serial Number and have the same telnet output as @michele8gr.
Flashing the Yi Dome 1080p (as it should normally be) is not working. Any pointers to a working solution is appreciated.

RonnyReporter, 2020-01-03:
Having there same output as @michele8gr. Same s/w and no firmware that seems to work on this camera. Hope someone finds a solution.

GilDev, 2020-02-14:
I tried resetting the camera by long pressing the reset button on the back before hearing the shutdown sound, but now it boots, makes it calibration and then makes 2 beeps continuously.
Have anyone seen that behavior before?

Happyboy0980, 2020-06-22:
Mmm how did you open telnet.. ? Seems on my version xy-r9820-k2 this is even blocked...

LuckyBenski, 2020-06-25:
> Mmm how did you open telnet.. ? Seems on my version xy-r9820-k2 this is even blocked...

I simply run `telnet 192.168.1.70` , no port details or anything.

RalfEngelberth, 2021-01-29:
Hallo, ich habe eine XY-R9820-G1 Mit der Firmware 7.1.00
Kann mir einer einen Tipp geben, welche Firmware hier funktioniert?
Über Telnet bekomme ich keine Verbindung.

RalfEngelberth, 2021-02-03:
Hello, I opened the camera and took a picture of the circuit board. Can someone give me a tip on how to hack or open RTSP?

[image]

[image]

rauko70, 2021-08-10:
Take a look to https://github.com/cjj25/Yi-RTS3903N-RTSPServer

cjj25, 2021-08-22:
> Hello, I opened the camera and took a picture of the circuit board. Can someone give me a tip on how to hack or open RTSP?
> 
> [image]
> 
> [image]

Thanks for the mention @rauko70 

@RalfEngelberth if you're interested in accessing UART on that MVT9632 then drop me a message :) 

The unpopulated header you see in the top left, it's the middle two pads.

I've managed to patch the uboot bootloader to enable UART for this board but annoyingly YI have used the RX/TX pins as dedicated GPIO too, so as soon as the driver loads.. no more UART. You can kill the init.sh before hand, useful for debugging nevertheless!

bondiblueos9, 2022-02-10:
@cjj25 how do you get the UART to work on the MVT9632?  I have a camera with the same board (not the same camera, in fact it uses tuya instead of yi) specifically labeled MVT9632-SC2303_QJ.V1.1 and I wired up those pads but got nothing.  I assumed from left to right they are ground, tx/rx, and 3.3v (measured), and I tried the tx and rx both ways.

cjj25, 2022-02-13:
> @cjj25 how do you get the UART to work on the MVT9632? I have a camera with the same board (not the same camera, in fact it uses tuya instead of yi) specifically labeled MVT9632-SC2303_QJ.V1.1 and I [...]

Check your two middle pins are connected to the RTS3903N chip using continuity mode on your multimeter.

UART_RX = 39
UART_TX = 38

Note: they might be the other way around, but they're the right pins!

Pin one is marked with the white triangle, using the image above, the pin numbers are going upwards.

[image]

If you need any more help, create an issue on https://github.com/cjj25/Yi-RTS3903N-RTSPServer