TheCrypt0/yi-hack-v4 #158: 15US yi dome 1080p

aladin2000, 2019-10-25:
hi There
thanks for your  work 

my camera is yi dome1080p
but 4 first digit are 15US
I wander if v4 is compatible 
and if contrôle directions are available

nicmorais, 2019-11-24:
According to the [wiki](https://github.com/TheCrypt0/yi-hack-v4/wiki/Supported-Camera-Models), this model is not supported yet.

goldbergr, 2021-01-13:
I have 3 Yi Dome 1080p with 15US flashed with yi_hack_v4_v0.2.0__yi_dome_1080p.zip from releases link.
The camera works from Yi Home app, and also works from homeassistant (with the ftpd fix of course)

aladin2000, 2021-01-13:
> I have 3 Yi Dome 1080p with 15US flashed with yi_hack_v4_v0.2.0__yi_dome_1080p.zip from releases link.
> The camera works from Yi Home app, and also works from homeassistant (with the ftpd fix of course)

what do you meant by ftp fix ?

goldbergr, 2021-01-13:
> what do you meant by ftp fix ?
https://github.com/shadow-1/yi-hack-v3/issues/129#issuecomment-361723075

aladin2000, 2021-01-21:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAD8AAAA/CAYAAABXXxDfAAAAAXNSR0IArs4c6QAAAARzQklUCAgICHwIZIgAAAAmSURBVGiB7cEBDQAAAMKg909tDwcUAAAAAAAAAAAAAAAAAAAAJwY+QwABivJx1AAAAABJRU5ErkJggg==###24x24:true###][Image de pistage][https://github.com/notifications/beacon/AD7K4I2TNFFYOVT4XIB6S2TSZXCI7A5CNFSM4JFBFDE2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFVC7YVQ.gif]

Seems that mine is broken

I was in yackv3 and I  Lost yihome connection

Then  i tried to recover but without success
Tried also to install hackv4 but the cam is not on router
I went back with recover but now

Always saying waiting to connect
Not able to make it work again on the ssid

Any Idea to un break ?