TheCrypt0/yi-hack-v4 #162: Cannot downgrade firmware once already updated to version 4.5.0.0c

pfeatherstone, 2019-11-21:
This is not really an issue with the repository but an issue in attempting to upload custom firmware when already at the latest version.
I upgraded my YI 1080p Home camera to version 4.5.0.0C_201910080934 and now cannot upload any other firmware. It seems it completely ignores what's on the SD card. Has anyone else ran into this problem?

skop1986, 2019-11-27:
I ran into the same problem

skop1986, 2019-11-27:
@pfeatherstone need to rename files in home_y203c and sys_y203c

silviob72, 2019-12-12:
> This is not really an issue with the repository but an issue in attempting to upload custom firmware when already at the latest version.
> I upgraded my YI 1080p Home camera to version [...]

Probably yo need this https://github.com/roleoroleo/yi-hack-MStar

ignusgraius, 2020-01-08:
> 
> Probably you need this https://github.com/roleoroleo/yi-hack-MStar

This worked for me. Thanks for the link!