TheCrypt0/yi-hack-v4 #163: Cacagoo wifi ip camera 1080 support

ricky0178, 2019-11-22:
Good evening,
I have this https://www.amazon.it/CACAGOO-Sorveglianza-Videocamera-Bidirezionale-Rilevamento/dp/B07WD8VC5M camera here. 
I was wondering if it was compatible with your firmware, and if yes what file should I use. 
It has an open port for telnet, it needs just the user (root)
if i run `cat /proc/cpuinfo` it displays me
`system type             : RLX Linux for IPCam Platform
machine                 : Unknown
processor               : 0
cpu model               : Taroko V0.2  FPU V0.1
BogoMIPS                : 497.66
tlb_entries             : 64
mips16 implemented      : yes
`

There is a file under /etc/version and it displays v2.3.2
`zhengqianbin@XY-201
2018-05-04 13:34:06
`
It can only be controlled by the YI-IoT app

thejanky, 2019-12-13:
Hi,
I got the same Camera.
Digging a bit deeper ( `cat /home/app/init.sh` ) I found that the Camera looks for a firmware file at `/tmp/sd/home_r10m` (or at `/overlay/home_r10m` )

Since this cam seems to be a re-branded version I hope it should be rather easy to create a working firmware.

RonnyReporter, 2020-01-03:
I have a slightly different camera but with the same s/w. 

~ # cat /etc/version 
v2.3.2
zhengqianbin@XY-201
2018-05-04 13:34:06
~ # 

Would be great to use this cam without the Yi app.

hyperbit00, 2020-02-19:
any solutions found for use without YI app?

adverserath, 2020-03-23:
also searching. Im wondering if we could use lede and reconfigure it. I think the board has serial connectors too

jcvillegasfernandez, 2020-03-23:
I have just this camera too.
Because telnet is activated, I had a look to the init.sh and there are some interesting lines 
   if [ -f "/tmp/sd/Factory/factory_test.sh" ]; then
     .....
     /tmp/sd/Factory/factory_test.sh
    exit
   fi
If I am not wrong, you could run a complete new program from the SD card using a personal "factory_test.sh".
Perhaps, adding in this file a new web page and an rtsp server and stopping the first script.
These things are beyond my knowledge and you will need Realtek RSDK-4.8.5p1 Build 2521 according to "cat /proc/version" But I could not find yet.

jcvillegasfernandez, 2020-03-23:
After more research, I have found something similar on GitHub  "fritz-smh/yi-hack" using a SD card and a "test" to bypass camera software. This hack includes an rstp server and a ntpclient to syncronize time, it could be a good base line.

adverserath, 2020-03-23:
lets have a look. is there any significance of the http.d file in /etc.
A:*   /secret:tonywu:123qwe   

There is no httpd services running, and only telnet is open

adverserath, 2020-03-24:
bricked it, screwed up the mount for home. 
Perhaps looking at http://www.live555.com/liveMedia/public/
ill see if serial works tomorrow.

jcvillegasfernandez, 2020-03-24:
You are right, there is no http server or nginx server commands, another task for RDSK.
If you have an SD card, have a look to the log file, there are interesting things and a lot of info.
I think without an RDSK we are stuck.
I hope you can unbrick it.

adverserath, 2020-03-24:
Not looking easy currently. Because there is no home, there is no init.sh. I'd probably need a full image to write over serial, which is one job to figure out. The other is getting hold of the image to write. I'm going to expect to hit a dead end with this. My other option is to use the case, and a raspberry pi zero, throw in a camera and wire up the IR sensor. Then I could just run motion

jcvillegasfernandez, 2020-03-24:
I do not know how to get the image file from the camera but you can wait until next firmware update, Ii could send you if they do not publish it.

adverserath, 2020-03-26:
I found a dump of the blocks, looking through that now. There doesnt seem to be a way to access the sd before mtdblock4, otherwise I could override the mounting point.

I just found this commented in mtdblock3/etc/init.d/rcS
#echo "enabling network devices ..."
ifconfig lo 127.0.0.1
#ifconfig eth0 172.21.210.225 netmask 255.255.255.0
#route add default gw 172.21.210.254
**#httpd -c /etc/httpd.conf -h /home/web**

adverserath, 2020-03-26:
Cant believe I missed it. There are TX and RX pins, the letters are so small so you cant see it. Trying to figure out the baudrate is a real nightmare currently. 
I think its running on 115200, but there must be other configurations.

jcvillegasfernandez, 2020-03-27:
As you said it is commented, and there are not any commands like httpd, rtsp, etc.
Looking other similar cameras the speed could be 115200 for serial connection.
Have a look on Internet I saw some places can help you with that (I don´t know where now) .

All IP cameras have similar software but for diferentes chips, so I think the software is prepared by the same company (maybe Yi Technology) and the version depends on the reseller requirements.
I found on Internet some cameras with the same Realtek chip as Cacagoo camera, but I could not find any firmware to download and extract those necessary commands.

adverserath, 2020-03-27:
Thought so. I tried 115200 with an rs232, but got garbage out. Unless my camera is completely bust. I'll try again later

jcvillegasfernandez, 2020-03-27:
If you get garbage I think it works, perhaps you mixed pins or level signal (I think it is 3.5 volts)
Perhaps you know more tha me but but just in case, first identify GND, after that play with one pin and GND if you get garbage is TX pin probably.
I wish you luck.

adverserath, 2020-03-27:
Pins are actually labelled on the board. I also get data on tx without connecting the 3.3v, just by grounding it. Ive only ever done this with arduino and routers  before, and I just google it enough to wing it.

jcvillegasfernandez, 2020-03-29:
Maybe the speed is 56700, I found this file /proc/cmdline:
console=ttyS1,57600 root=/dev/mtdblock3 rts_hconf.hconf_mtd_idx=1 rts-quadspi.c
hannels=dual mtdparts=m25p80:8192k@0(global),192k@0k(boot),1344k@192k(kernel),1
792k@1536k(rootfs),3072k@3328k(userdata),1664k@6400k(rootfs_backup),64k@8064k(v
d1),64k@8128k(conf)

adverserath, 2020-03-29:
Will check. Thanks.

jcvillegasfernandez, 2020-03-30:
I could get images from my system if you are interested I could send you them.
It was easier than I thought (using dd if=/dev/mtdblockx of=mtdx)
/dev/mtdblock0 Whole memory 8Mb, complete chip m25p80 
/dev/mtdblock1 Boot partition 
/dev/mtdblock2 Kernel partition 
/dev/mtdblock3 Root partition 
/dev/mtdblock4 User partition
/dev/mtdblock5 Backup partition
/dev/mtdblock6 vd1partition, ¿video overlay?
/dev/mtdblock7 conf partition, personal user data, AP SSID, password, keys

adverserath, 2020-03-30:
Sure thing. Just remember to remove your ssid before you post it 😉

vinceHH, 2020-03-31:
Hello, it seems to be the same hardware than the Yi IOT outdoor camera, I'm looking for a solution to hack this camera. thanks for your update :)

tdiz, 2020-04-04:
hi, some questions
I  have got images (by using dd if=/dev/mtdblockx of=mtdx). Does it enough to repair cam in case of mistake during experiments?

My camera looks like [this]( https://www.cacagoo.com/p-s3341eu.html) but labeled as YI .

jcvillegasfernandez, 2020-04-06:
"tdiz", no sure 100%  if it works in the reverse way dd if=mtdx_file of=/dev/mtdblockx.
I didn't try yet.
But if this does not work I am sure we can find any "mtd" commands to write back to the m25p80 chip. If I have time I will try to figure it out, I keep my camera in the box ;)

jcvillegasfernandez, 2020-04-06:
"tdiz", I have just unbox my camera to check your question.

First I tried the reverse way dd if=mtd7 of=7dev/mtdblock7 (user area because it can be reset by button if something wrong happen), and it gives me no errors but I am not sure if this did the job.
Looking for another command I found in the backup directory a command "mtd_img", use mtd_img mtd_number image_file"
I tried again with user area to be in the safe side.
    /backup/mtd_img 7 mtd7
Output:    
    mtd name: conf
    total blocks 1, erase block   1/  1
    writing image file....

and it seems to work after switch the camera off and on again.
If you need to recover your camera first you could try this mtd_img command, but I would save a copy of this command in the sd card first.

manu-971, 2020-04-22:
Hello ! I'm looking also for a hack of this camera.
My IP camera is not under the Cacagoo brand but it is exactly the same (same /proc/cpuinfo and also /etc/version, as well as the home_r10m file received for a firmware update).
FYI, on the packaging, I see as reference PP2422 or XY-R9820-F4. It says Smart Camera (probably the name of the model) and IOT Cloud (probably the brand).

zackfuchtel, 2021-03-23:
anybody something new? thx

ricky0178, 2021-03-23:
I don't think so, sorry

wooshy1, 2021-05-15:
ok so this camera shares the same chipset as the tplink tapo c200 if you go to the tplink gpl source code website you can download the source for that camera which contains the sdk required.

cjj25, 2022-01-06:
https://github.com/cjj25/Yi-RTS3903N-RTSPServer

sederedelcane, 2025-05-11:
Hi, are there any news about hacking r10m firmware? Thanks