TheCrypt0/yi-hack-v4 #165: Support for KEYKE

grausof, 2019-12-02:
there is a new camera (KEYKE) that appears to be a clone of the YI Outdor. This is the [amazon link](https://www.amazon.it/Telecamera-Videocamera-Sorveglianza-Movimento-Archiviazione/dp/B07SC9MKK6/ref=sr_1_4?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&keywords=keyke&qid=1575318121&sr=8-4). Has anyone tried to change the firmware?

cook32, 2019-12-28:
I have same camera but firmwares are not working.
I hope somebody will find something. It is really big need to have RTSP server...

grausof, 2019-12-28:
> I have same camera but firmwares are not working.
> I hope somebody will find something. It is really big need to have RTSP server...

### 💪🏻Enable TELNET and FTP !
I started getting my hands on this new camera, and for now I have managed to enable telnet and ftp. You can follow my [guide](https://github.com/grausof/keyke-hack). The camera is based on MStar Infinity3 chipset.I think it's possible to use some of the work done for other cameras to enable rtsp on this too.