TheCrypt0/yi-hack-v4 #166: No access to #supporters at discord, no licenses, no response ...

not-so-important, 2019-12-03:
Hi,

Trying to get a reaction from the Author (@TheCrypt0), although my request is slightly "off topic" for the core GitHub project itself.

The needed rtspv4 and viewd are excluded and monetized (viewd being donationware) via the discord entity, which is in general a valid method to fund a project.

What's not o.k. is that after the mandatory donation (in my case roughly 25€ for 4 licenses/cameras) there is just absolutely nothing happening, in my case since a week by now ... and as of some messages in the discord channels, I am not the only one.
 
- No access to the #supporters channel to download the two binaries.
- No licenses (although announced to be delivered within 24 hours in the confirmation email about  the donation having arrived).
- No response on any communication whatsoever (discord chat, direct email) 

As said, slightly off topic for the project itself.

To @TheCrypt0: Making a donation per license (in this case per camera) mandatory is very close to requesting a commercial license fee, which sets an expectation on an at least minimal service level to be met.

TheCrypt0, 2019-12-03:
Hi @not-so-important,
Thank you for the report. I'm aware of the situation and I'm working hard to handle all the requests.

* The licenses are automatically generated once per day, did you check the spam folder for them?
* The viewd file is linked on the license's mail in case the #supporters access wasn't given by Integromat.
* I have a queue of messages to reply to and I read all of them. This week it exploded because the automatic service stopped working (it was being rate limited by Discord) and the time needed to catch up with all of them increased exponentially.

I'm currently catching up with all the requests and handling the situation as fast as I can, the #supporters access should have been given to by now.

Let me know!

not-so-important, 2019-12-03:
Thanks for solving the case ... got my cameras running as expected now ... great project!

closed 2019-12-03 by TheCrypt0