TheCrypt0/yi-hack-v4 #173: High CPU usage even with everything disabled

codeclinic, 2020-01-03:
I'm running Hackv4 on a Yi Camera Outdoor 1080 with all functions disabled other than the required HTTPD. The cpu loads are typically like so:

4.08 3.31 1.74

Do we have any idea how many cores this device has? Even if it's running on a quad-core (unlikely) the cpu loads seem high for next to no services running. 

With RTSP enabled it's only possible to run the low res encoding, and then the stream is approximately 8seconds delayed. With HQ stream and no low res stream the stream is unavailable.

If I enabled only HTTPD, HD Stream, & MQTT the stream becomes unavailable, the web interface is almost always timesout and nothing is sent to MQTT broker. On the rare occasions that I can get into the web interface it shows cpu load averages of 5+ explains the unresponsiveness.

This unfortunately makes the camera essentially unusable in anything but a very basic low res delayed feed. 

Is this a know issue? Any suggestions for making performance better?

tiredjohn, 2020-01-11:
Same here, I have the same camera and everything disabled except for HQ RTSP. My server load is at 6.29 6.04 6.11 - it's frequently unresponsive and sometimes requires a hard reboot.

codeclinic, 2020-01-12:
I can't even run it on HD stream with everything else disabled. If I try HD stream it essentially kills tje device. I get no stream output and it takes hours of trying before I can log back in to the web interface in order to switch back to low res rtsp stream. As it is low res barely works, it's choppy with lots of noise.

grungeguitarist, 2020-01-17:
This problem really keeps me from using the hack on my other cameras.

codeclinic, 2020-02-11:
From what I can gather this uses an ARM Cortex v9 which at best has 4 cores. So this suggests that I am correct in thinking that the device really is being overowrked just by running this hack in its most minimal form. Essentially making it incompatible, unless the code can be revised to use less cpu time.

lucaventurini, 2020-02-11:
Hi, I have this issue #88 since I installed the hack. Do you think it might be related to the CPU usage? Do you experience the same?

codeclinic, 2020-02-11:
It could well be the cause.

dj-matt-e, 2020-09-08:
I don't know if it's related to your problem but I've also had bad performance from my Yi 1080p Outdoor cam. I always assumed it was because the 2.4GHz signal was weak at the front of the house. I installed a booster but yet I still had problems with it cutting out after about 5 seconds and struggling to scrub through footage. Also when trying to use the FTP to transfer some files it would start off OK and then hit a bottleneck where the speed of transfer would dramatically drop. Today I stumbled upon the solution. The USB plug I had powering it was an old iPhone 5V 1A supply. I swapped it today for a 5V 2.4A supply and suddenly my issues have gone. It seems like the 1A supply was under powering the camera the whole time.

lucaventurini, 2020-09-08:
I instead have solved my issues #88 by removing the SD card (and giving up on recording). Now the video stream works fine.

TheCrypt0, 2020-09-08:
Hi @lucaventurini thank you for your suggestion!
May I ask what camera model did you test the fix on? I might be caused by a slow SD card or something I'm not yet aware of.

lucaventurini, 2020-10-12:
Hi @TheCrypt0 I have a Yi Home, with firmware 1.8.7.0F etc.

kabakakao, 2024-10-23:
any news on this issue?