TheCrypt0/yi-hack-v4 #176: YI Cloud Dome Camera, MicroSD, ETH port

MasterXur, 2020-01-16:
I`ve installed yi-hack-v4 successfully. 
Settings for Micro SD card is not present on Yi Home App settings page, in config Recording without Cloud is turned on.
My cam is model **8FUSY04WJZ.**

Also, is there a way to make a selection for type of the interface used, WiFi or Ethernet...?

When on Wifi MAC is 0C:xx:xx:xx:xx:x, on ETH MAC is D2:xx:xx:xx:xx:x (xx is the same MAC address on both port), 
Different IP address were used by camera 192.168.1.13 for WiFi, 192.168.1.14 for ETH, depending with interface is used.
WiFi is made by SHENZHEN BILIAN ELECTRONIC CO.LTD, ETH module is unknown.

Thanks

AndikaTrisna, 2020-10-03:
Hi helo,
Could you confirm that ethernet port is working in this hack?
I am considering to buy this camera, but not sure if the ethernet port is working. 

It will be great if we can choose which interface to listen.

MasterXur, 2020-10-29:
> 
> 
> Hi helo,
> Could you confirm that ethernet port is working in this hack?
> I am considering to buy this camera, but not sure if the ethernet port is working.
> 
> It will be great if we can choose which interface to listen.

Sorry for late reply,
ETH port is working with this hack. Both of my cam are with same hardware, so they have two independent IP address.
But there is no interest for developing SW switch for choosing with interface will be primary.