TheCrypt0/yi-hack-v4 #179: Can't flash - nothing happens

sekkr1, 2020-01-19:
Hello,

I have a Yi IoT 1080p outdoor camera (model XY-R9520-V3) and nothing happens when I turn it on with the SD card in (16gb capacity fat32).

The camera would just turn on regularly doing its sounds and no LED would blink, just the regular red LED lighting up and letting it sit for a couple minutes and trying to access the web panel shows that it really didn't flash.

I tried flashing before and after connecting to the Yi IoT app to no avail.

Is it really just opening the side panel, inserting an SD card with 2 files in and connecting the camera to power?

Maybe it doesn't work because it's an "IoT" camera?

I'd appreciate it if someone could help me, thanks.

maartenvang, 2020-01-25:
I have the exact same issue as you.

- Yi Home 1080p (YYS2016)
- 16 GB SD card, formatted as fat32
- The correct files for my camera on the SD card

I tried formatting the card with my macbook first, but with ubuntu and gparted didn't work either. I even tried a smaller partition. All that happens is the regular startup flow (solid orange light, then a shutter click, then the light goes out and I can access the camera as usual).

I suspect the SD card itself is the culprit, I'll have to try another one but I don't have one at hand.

jonydebu, 2020-01-31:
Same issue here .

ebolisa, 2020-02-03:
Same issue here too.

Peboli, 2020-02-03:
Same issue with a Yi Home 1080p (original firmware 4.5.0.0C_20190080934)

stixpunk, 2020-02-09:
Same here. I sucessfuly flash Outdoor Camera 1080p, but nothing happened on 1080p Home. Used same sd card  which work on outdoor, files and names are correct. After put card in and turn on camera, it runs normally and start recording on sd card.
Product name: YI 1080p Home Camera
Model: YYS.2016
Original fw: 4.5.0.0C_201910080934 (on previous version nothing happened too)

ndonna04, 2020-02-12:
Same here. Yi Home 1080p (original firmware 4.5.0.0C_20190080934).

stixpunk, 2020-02-12:
Ok, I'm found, that some newer Yi Home 1080p start using another chip. If crypt0 version don't start flashing, try use this - https://github.com/roleoroleo/yi-hack-MStar

supermanscottbristol, 2020-02-19:
Same here. Yi Home 1080p (original firmware 4.5.0.0C_20190080934).

benogen, 2020-03-20:
Anyone able to compile something similar for the Yi IOT outdoor camera (model XY-R9520-V3)? Or does this already exist somewhere?

JhonnyMarr, 2020-05-09:
I´m also interested in yi_Hack_v4 for the  Yi IOT outdoor camera (model XY-R9520-V3).
I´ve got one of them. I´ve seen that the telnet port is open (user: root no passwd)
If any coder needs something tell me what to do.
Regards

Abuelika, 2020-05-12:
Is the port open? It tested on port 23 and is closed.

JhonnyMarr, 2020-05-12:
In my camera the port is open

Abuelika, 2020-05-13:
Hi,
I have been looking here, (https://github.com/rage2dev/YiOutdoor)
And I've got telnet access on the camera on port 9999
Tnx

JhonnyMarr, 2020-05-13:
My firmware version is Firmware version 7.1.00 11A201905311440  and it has port 23 open.
I´m very interested in activate rtsp. If anyone can help woud be very appreciated
~ # cat /proc/cpuinfo                                                                                                   
system type             : RLX Linux for IPCam Platform                                                                  machine                 : Unknown                                                                                       
processor               : 0                                                                                             
cpu model               : Taroko V0.2  FPU V0.1                                                                         BogoMIPS                : 497.66                                                                                        
tlb_entries             : 64                                                                                            
mips16 implemented      : yes                             

There is another thread about it here https://github.com/TheCrypt0/yi-hack-v4/issues/226

ilmisu, 2020-06-01:
Hi all, same issue on my Yi Home camera 1080p BFUS serial...

Tried with 4 or 5 different SDs of different capacities, but every time it did its boot with never check out for the microsd. 
Wait for an help... 😭

llellouche, 2020-06-01:
Yi 1080p Home 4FUS Firmware 4.2.0 doestn't works too.

I juts bought this camera it's new model 3 of Yi 1080p Home

stefankoehne, 2021-01-22:
> Hi all, same issue on my Yi Home camera 1080p BFUS serial...
> 
> Tried with 4 or 5 different SDs of different capacities, but every time it did its boot with never check out for the microsd.
> Wait for an help... 😭

Try this. https://github.com/roleoroleo/yi-hack-Allwinner-v2
This worked for me for YYS2016 - BFUS - Firmware 9.0.19*

git-voopeel, 2021-03-19:
Hi all, I cant still upgrade my Yi Home 1080p BFUS serial number. I search for Allwinner v2 but ... there is no binary files like in other repositories. Shall I copy all file tree? Thank you for help.

Abuelika, 2021-08-18:
Hi,
I have tried the solution of this link and it has worked for me.

https://github.com/cjj25/Yi-RTS3903N-RTSPServer

I have connected the camera by rtsp with zoneminder without problems.

YI IOT OUTDOOR CAMERA XY-R9520-V3