TheCrypt0/yi-hack-v4 #188: YI IOT Outdoor Camera supported?

thrasheri, 2020-02-16:
Hi! 

Im looking for a new outdoor camera to use with home assistant. I noticed there are 
YI Outdoor camera and YI IOT Outdoor camera. Is the YI IOT Outdoor camera supported? The Iot version seems to be cheaper.

majkers, 2020-02-24:
Same for me.

viorel-anghel, 2020-03-18:
nope, yi iot outdoor is not working with yi-hack-v4. 
i have both yi outdoor 1080p which worked and yi iot which hasn't. 
basically it just not updating/reflashing the system and it is running the default xiaomi software.

muhkuhgh, 2020-04-07:
Any news on this? I recently failed for one too.

fredwise, 2020-06-13:
Hi! Also looking for support for the Yi IOY Outdoor Camera.

alfem, 2020-07-16:
Me too. I bought it because I thought the hack existed. But I did not notice the "Iot" model in the Yee Camera description.

moshik132, 2020-08-10:
I also have a camera YI IOT OUTDOOR
Is there a date when it will support?

djscolla, 2024-10-26:
Hi! Also looking for support for the Yi IOY Outdoor Camera.