TheCrypt0/yi-hack-v4 #189: qnap surveillance station

resheph73, 2020-02-17:
Did anybody manage to properly connect a yihack'ed camera to qnap surveillance station? I have a dome camera 1080 and the rtsp stream works properly when opened via VLC but even tough the test connection is successful the stream in SS is always black and the link stays "inactive". 

Any idea?

davide-rhw, 2020-02-21:
Same problem, i try to open a ticket to qnap

paulioliolio, 2020-04-08:
You'll  need to add the camera as a Generic RTSP camera. Populate the IP field with the IP address of the camera, and add the ending of the URL for the (high or low quality) stream you want. This works for me in that I can monitor the camera from Surveillance System, but I am unable to activate alarm recordings. Please let me know if you have success with that, as well as getting the camera set up on your system. My config is attached (I'm using the MStar Yihack but I imagine its the same).
[image]

davide-rhw, 2020-04-08:
Hi, what is the name and password? the yihack 4 not have user and password for stream

Thanks

paulioliolio, 2020-04-08:
I have enabled the username and password in my yhack settings, but you're right I do not think it is mandatory or enabled by default.

paulioliolio, 2020-04-08:
By the way, QNAP came back to me after I raised a ticket. They said that alarm recording is not possible when using the generic RTSP device in Surveillance Station. They said it is only possible for an unsupported device which can be setup as an ONVIF profile S device, sadly I had no success with this on my yihack.

davide-rhw, 2020-04-08:
Where i enable the username and password?

paulioliolio, 2020-04-08:
in your yhack config

alienatedsec, 2020-10-20:
I still have issues with yi-hack-v4 - I have two cameras with the same problem.

[image]

Also, it says idle whenever connected. Does anyone have any ideas on how to fix this?

[image]