TheCrypt0/yi-hack-v4 #190: Anyone got a license lately?

adam1942, 2020-02-18:
Discord seems full of people who have donated but got nothing in return.

TheCrypt0, 2020-02-19:
Hi Adam,
I'm still generating the licenses once per day, If you haven't received yours it might be in the spam folder.

I'm overwhelmed by mails and messages, unfortunately it takes some time to catch up with all of them and I haven't been able to follow Discord because of this. I'm trying to keep a steady rate on the email's reply.

**To everyone reading this: If you haven't received your license yet, PLEASE reply to the mail I sent you with the details of the request. That's the only way I can get to you without getting blocked.
The providers have flagged my personal email as spam due to the license file attached and I'm having troubles reaching you.**

The following is from one of the Supporters which didn't get the mail:
> [...] the email is flagged by Microsoft threat detection as bad and the email was sent to quarantine without any notification and no record in spam folder

or another:

> Yes sorry, I just checked my spam folder as you suggested. They are there.
> Thank you!

Some providers directly reply with:

> The response from the remote server was:
> 550 Spam message rejected

But unfortunately most of them don't, I have no way of knowing if you got your license or not.

Let me know if you have any suggestions, or more questions.

Crypto

adam1942, 2020-02-19:
Thanks Davide. I can confirm I got my license - Much appreciated :) - I'll share on discord also.

TheCrypt0, 2020-02-19:
Thank you Adam!

I'm keeping this open so more people will see it and it will provide more clarity about the issue.

adam1942, 2020-02-19:
Hi Davide,

Just a though - Citrix Sharefile offers a "notify on download". If you
where to use Citrix Sharefile or equivalent you could create a folder and
share that folder with the client (locked to their email address). Each
time its downloaded or accessed you can be sent notification. It would mean
a bit more admin work for creating the shares, notifications etc and then
also some way of correlating the data in excel regarding issued vs
downloaded/consumed.

(I've included a screenshot of the "admin" screen below).

Either way good luck and thanks for coming back to me.

Cheers,
Adam.

[image: image.png]

leroynobel, 2020-02-26:
Crypto,

I donated 6 days ago. but do not have access to the supporters. I've checked my spam, but haven't received anything yet. can you check this! 

my account midget is # 3214

 if you need more information, please let me know.

TheCrypt0, 2020-02-26:
Hi,
Did you use the donation link provided by the Discord bot? I can't find
your request.

Let me know.

>

leroynobel, 2020-02-26:
thank you for your reaction,

Yes I have used the link see attached screenshots here.

Please look forward to your response.

TheCrypt0, 2020-02-26:
@leroynobel 
I think the screenshots got missing while posting the reply. We can keep in touch directly via mail, you can find mine in the PayPal transaction mail. 😄

leroynobel, 2020-02-26:
Hi

herewith again the attachment of the transaction that I submitted.

Mijn account : midget#3214

Please see your response

leroynobel, 2020-02-26:
did you receive my mail ? @TheCrypt0

TheCrypt0, 2020-02-27:
Yep, I've just replied.

>

leroynobel, 2020-02-27:
I've sent you the requested information, @TheCrypt0

tnr20, 2020-03-20:
i have payed last year in August and wait still for my License!

Transaktionscode
 8LV13299UU8391831

Can you give me now my license?  i have waited for more than 1 year and you have never answered my questions in Discord

mcgurdan, 2020-03-29:
Can I get a licence please? I am unable to Donate via Paypal: https://github.com/TheCrypt0/yi-hack-v4/issues/212

TheCrypt0, 2020-03-31:
Hi @tnr20 
That's weird, I've never banned anyone for asking for a license or info. I just checked in the server's ban list and you are not there. I'll send a ticket to Discord to get some clarification.

Anyway if you could drop me a message on my personal mail we'll handle the issue there 😄 

---

@mcgurdan 
Thank you for letting me know. I'm replying in the issue https://github.com/TheCrypt0/yi-hack-v4/issues/212 thread.

alienatedsec, 2020-11-24:
> Hi Adam,
> I'm still generating the licenses once per day, If you haven't received yours it might be in the spam folder.
>
> I'm overwhelmed by mails and messages, unfortunately it takes some time to [...]

You still have not caught since February. I am starting to regret in supporting your business model ;)

TheCrypt0, 2020-11-24:
@alienatedsec 
Are you referring to licenses? If so, please feel free to drop me a mail and I'll be happy to get back to you as soon as I can.

alienatedsec, 2020-11-25:
@TheCrypt0 it depends if you are planning on further development? Other projects are being updated regularly

tnr20, 2020-12-14:
i have buyed long time ago via paypal but i have never received a license key. i said you that i not received a key for the 4th times now.
Why do you NOT give me my License Key? or send my money back. my paypal adress is: emine.tek85@gmail.com

Please send me my License Key. I asked you via discord with this email or 5dl@protonmail or tnr@protonmail and you have banned me
thanks Sir!

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

TheCrypt0, 2020-12-15:
@tnr20 
I have never banned anyone from Discord, especially the ones requesting what they paid for. I'm replying to you privately and I'm sure we'll sort the issue in a couple of minutes, keep an eye on your inbox 👍

damounh, 2022-01-06:
@TheCrypt0 seems like many people are still waiting for a license on the discord channel. I've just sent a request for a license. Are you still supporting this project? Looks like there has not been any activity for more than a year.

tnr20, 2022-01-10:
i newer become my License , because i asked a question in Discord and after that i was banned. this was 2-3 Years ago!
Like i said if you dont give me my License than give me my money! unbelivabaile how much times i said you that. look on paypal!
MY PayPal Email Adress was: ***@***.***

Like i said it 4 times bevore! i wait maybe over 4or 5 Years for the License. . in my previous emails i give him all the details paypal transaction and so on.
but it seems that you have scammed me. dont forget Karma my friend! all the good things you do , come to you back one day. And all the bad things you do in this world come back to you again!
maybe not today, tomorrow or next month... but it comes! may in 3weeks? or 3months? or 3Years! or in 5Years but it comes.... i dont want my money. Leech it..
Karma will do the job for me and all other people that has bad things do to me. Wait and see.. I dont want you , to make fear!!
but this is World and Karma is the Balance... i wish all the best to you.
Best Regards

Sent with [ProtonMail](https://protonmail.com/) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

alienatedsec, 2022-09-30:
I can't believe people still buying a license for `yi-hack-v4`
I bought myself in the past, and the project died. Look at the below and contribute to a good cause if you want to spend money on something.

[https://github.com/alienatedsec/yi-hack-v5/](url)