TheCrypt0/yi-hack-v4 #192: Yi 1080 Home firmware cannot be flashed

afanhk3, 2020-02-20:
I got the Yi 1080 home from Taobao China, but I tried copying various version of firmware into the MicroSD for flashing .  However none of these files trigger the flashing.  The camera keep repeating waiting for connections.  

rootfs_y20  home_y20
rootfs_y18  home_y18
home_y203c  sys_y203c

WattsC-90, 2020-02-20:
Same problem for me:(

starbuck93, 2020-02-22:
Same here. Yi 1080P Home Camera. Model number on the box says YYS.2016 fwiw. Just got this new from [Amazon ](https://smile.amazon.com/gp/product/B01CW4AR9K)today.

bridgemode, 2020-02-24:
The brand new Yi cameras seem to have the QG2101 chip, which is developed by Qigan Technology (a subsidiary of Yi themselves). I personally don't know if anyone has been able to get the code from these chips yet

ijalalfrz, 2020-02-27:
Same here i got YYS 2016 model

Brainz78, 2020-02-29:
Just purchased a Yi Home 1080P from UK Amazon after finding this project while researching ZoneMinder compatible cameras. It arrived with Firmware 4.5.0.0C_201910080934, followed all instructions to the letter multiple times with various 8Gb SD cards but it will not load the custom firmware :-(

On resetting the camera the light goes steady yellow for 16 seconds then into the stock 'waiting to connect...' loop, it looks like Yi may have disabled the loading of firmware from the SD card or added some other type of verification.

Hoping this is possible to fix as this is a great project I will happily support if I am able to use.

rralleres, 2020-02-29:
Using custom firmware for MStar platforms worked for me (see https://github.com/roleoroleo/yi-hack-MStar). I have a Yi Home 1080p camera (YYS.2016 model) with 4.5.0.0C firmware

Brainz78, 2020-02-29:
thanks @rralleres that worked a treat for me - well what do you know, the manufacturer went and switched the chipset :-/

WattsC-90, 2020-02-29:
@rralleres Thanks for the tip, tried that just now and ignores it totally, "waiting to connect".. *sigh*.

I have the 9FUS camera (from the serial on the back), am I looking at the wrong thing? kinda sucks that I cant configure my own rtsp streams!

starbuck93, 2020-02-29:
@WattsC-90 do you know which firmware you have? https://github.com/roleoroleo/yi-hack-MStar/issues/91 Mine's 8.2.0x so I'm having the same issue.

WattsC-90, 2020-02-29:
@starbuck93 8.2.0.0A.. wonder if theres an older version I could downgrade to that would make this work..

wadalino, 2020-05-14:
Any news about chipset QG2101A ?
[image]

ethempekin, 2020-07-29:
any update on this issue?

BerlinChan, 2020-09-15:
> @WattsC-90 do you know which firmware you have? [roleoroleo/yi-hack-MStar#91](https://github.com/roleoroleo/yi-hack-MStar/issues/91) Mine's 8.2.0x so I'm having the same issue.

Try this: https://github.com/roleoroleo/yi-hack-Allwinner

My Yi 1080 home / YYS.2016.CN / version 8.2.0A_201912270941, can flash with this, and access web interface at port 8080

mianrelie, 2020-10-28:
> Try this: https://github.com/roleoroleo/yi-hack-Allwinner
>
> My Yi 1080 home / YYS.2016.CN / version 8.2.0A_201912270941, can flash with this, and access web interface at port 8080

It work! Thanks!
Yi 1080p Home
BFUS
Model: YYS.2016
Firmware: 8.2.0.0A_202007211802

avellma, 2022-08-28:
I have the same problem and tried everything above..... Please Somebody help me.... I really really want to make this camera work.

My camera is:
- Yi 1080 home 
- Model Number: YYS.2016.CN 
- Firmware version: 11.1.35.3_202207191613

avellma, 2022-08-29:
Here I am adding more information about my calera

> I have the same problem and tried everything above..... Please Somebody help me.... I really really want to make this camera work.
> 
> My camera is:
> 
> * Yi 1080 home
> * Model Number: YYS.2016.CN
> * Firmware version: 11.1.35.3_202207191613
> 
> I added to the SD Card the following [image] The SD card is a 16gb and formatted as recommended SD Card as FAT32

Here I am adding more information about my camera and what I have try to do....
I did try the yi-hack-Allwinner-v2, but it did not work. I think the firmware in allwinner is up to 8.xxx.xxx.x

My camera is:

Yi 1080 home
Model Number: YYS.2016.CN
SN: RY291GUSWZ27004047
Firmware version: 11.1.35.3_202207191613
The SD card is a 16gb and formatted as recommended SD Card as FAT32
So I used the y291ga ... not sure if it is the right folder...
I copied the following files to the root of the SD Card .....
|-- Factory/
|-- yi-hack/
|-- lower_half_init.sh

Definitely I need help as I am not sure on what I am doing.
I really really really would like to hack this camera if it is available.