TheCrypt0/yi-hack-v4 #197: Outdoor camera FW can´t be flashed

HR787, 2020-02-29:
Hi, i have bought a "YI" outdoor camera. The camera use the YI IoT iOS App and has the FW 7.0.01.20A_201905161919. Do i have any chance to flash the FW? I think the camera is abolutely identically constructed to Outdoor H30. I would like to use a RSTP Stream... Thank you in advance for helpful answers.

robmcc1983, 2020-03-18:
Hi, I have the same camera as you.
Did you ever get a answer or try this as I would like to know if this would work. 
Thanks

tijo82, 2020-03-19:
Anyone got this cam working? I think I have the same issue (XY-R9520-V3)

benogen, 2020-03-20:
There is a FW for the newer home Yi cameras https://github.com/roleoroleo/yi-hack-MStar They have another chip. Anyone able to compile something for our model?

alfem, 2020-03-30:
I also got a Yi IOT Outdoor Camera and it does not load TheCrypt0 firmware. 

I can telnet to port 23 with root (no password), and read some logs, but nothing else. 

/proc/cpuinfo is like this:
```
system type             : RLX Linux for IPCam Platform
machine                 : Unknown
processor               : 0
cpu model               : Taroko V0.2  FPU V0.1
BogoMIPS                : 497.66
tlb_entries             : 64
mips16 implemented      : yes

```

vinceHH, 2020-03-31:
Same model here and same issue :(
Did you know if this model will be supported ?

stevedee78, 2020-04-03:
I have the same Yi IOT Outdoor Camera 1080 it is impossible to flash.

HR787, 2020-05-06:
Any update for the hardware/firmware?

moshik132, 2020-08-10:
Any update for the hardware/firmware?

alfem, 2023-03-16:
I think the only project focused on our _Yii IOT outdoor camera_ is this one: https://github.com/cjj25/Yi-RTS3903N-RTSPServer