TheCrypt0/yi-hack-v4 #200: Shareware

mclaughlinhugh4, 2020-03-02:
Please make it more clear that this needs to be paid for for it to be useful. Unbelievable waste of my time. Its one thing hacking hardware, but charging for hacked software (based on firmware that you don't own) is crazy! I appreciate its a lot of work, so do Accept donations, but don't lock down the software.

TheCrypt0, 2020-03-03:
Hi,
The firmware is open source and available for everyone to use, it unlocks
most of useful and hidden features of the camera as well as disabling the
communication to external server.
The only part of it that is locked down is the h264 grabber from the memory
(which is software that I wrote from scratch since it didn't exist), it
took me hundreds of hours to make it work and without donations I couldn't
reserve this much time for the project (which of course includes all the
other parts of the firmware).

It would be silly to charge for something I didn't own, you are free to use
the firmware without any charge and even contribute to the project if you
want!

Let me know if you have more questions.
Cheers,
Crypto

ranrinc, 2020-04-09:
> Hi, The firmware is open source and available for everyone to use, it unlocks most of useful and hidden features of the camera as well as disabling the communication to external server. The only [...]

Agree the hack is 100% great... now if only @TheCrypt0 willing to add more camera... my new yi 1080 does not supported so I will have to use mystar hack.. but the cam quality is not up to yours..

bytesex, 2020-05-04:
My comments, @Mullen47 
1. I have to pay for the every device I own. That's a shareware, not a donation at all. Why dont I able to use the piece of software I already paid for on all my devices?
2. During "donate" submission there is statement like "If you have 3+ devices, contact with me on Discord and I will give you a discount:)" 
   Isnt it something wrong with "donate" and "discount" used together?
3. The author comments: "it took me hundreds of hours to make it work and without donations I couldn't reserve this much time for the project"
Well, the first comment was

https://github.com/shadow-1/yi-hack-v3/issues/126#issuecomment-449659465
TheCrypt0 commented on 23 Dec 2018 
"Hi everyone, after some work I managed to use the SDK to compile @dvv's grab to test it on my Yi Dome 720p camera."

The last was:
https://github.com/shadow-1/yi-hack-v3/issues/126#issuecomment-454068318
TheCrypt0 commented on 14 Jan 2019

"I tested the above process with a small program I wrote, just to be sure that my
    assumptions were correct. Just redirecting the output of it to a file *.h264 worked
    like a charm."
  
It took 20 days for the author to compile something workable based on @dvv's code. Soooo big efforts -lets calculate further.
Even if the author worked 50% of this time, he could spend 20/2*24 = 240 hours

Based on Discord channel stats, he already issued 3000+ licenses. Each license costs a "donation" of 5pounds and higher - you have a choice to pay more.
So, the author earned at least 15.000GBP = 17000EUR = 18600USD
His big efforts were worth, respectively, 62.5GBP/hour, 71EUR/hour, 77.5USD/hour. High skilled IT coder doesnt earn event part of it.
Observing current issues indicates he doesnt support his project, and the latest release was an year ago:
(0.2.0 TheCrypt0 released this on 6 May 2019). BUT his Discord channel looks like the store and gets 10-15 new users every day requesting the licenses.

4. Because the viewd and rtspv4 are not open-source, its not guaranteed they are free from vulnerabilities and hidden functionality. What if the camera sends some data outside of your network?
Using Xiaomi implementation, I am sure their soft have passed some external audit, but what do we have here? What if there is some eavesdropping functionality, spying you voice and video activity?

5. His license ordering based on jotform.com forms, which offers PayPal integration for 40-50USD/month. Will someone use paid platform only to collect donate? Only if you have a treasure chest getting you with a constant cash.

So, it seems its 17-year old enterpreneur just happy to sold the software mostly based on other's work. He just makes money out from the small piece of code, discontinuing the project support and complaining the lack of time. I am also guessing he doesnt pay taxes, because it is a "donation" (I am not sure, not a lawyer here).
But still having time to manually send the licenses, though. 

@TheCrypt0 I know about "mind you own business", but.. Either sold and support the software without using "donation", or admit your efforts are completely compensated with 17000eur of revenue and do not lock the codebase.

TheCrypt0, 2020-05-04:
Hi @bytesex,
Thank you for taking time to track down all of my progress with `viewd`. Unfortunately on GitHub there is about 5/10% of what I've done to make it work. You are missing 2 complete re-writes, reverse engineering, and the firmware development.

First, you are right. It took me a relatively small time to make the the very first version of the software. It was super buggy and the connection was dropping continuously.

The problem was the approach to the problem.

I tired to follow dvv's ideas to continue developing it but without much success, the stream worked but it was super glitchy and unstable. The only way to get _something_ out of the camera was using VLC (with a lot of errors), an NVR like Shinobi would not accept the video stream at all.
At this moment people were absolutely free to donate whatever amount they wanted, there was no lower limit, people even donated 10 cents and I gave them access to the beta builds. The donation was symbolic.

Weeks pass and I try another approach with a complete re-write of the analysis engine of the circular buffer. It would search for buffer overwrites (it's a circular one, remember), find malformed H264 NAL units and fix them, I tried to patch Xiaomi's encoder to make it spit out working video frames but without much success. The connection was more stable but random hangs of the RTSP server would require a reboot of the camera and they were not detectable by the software. Of course this was a no-no for security reasons. This was the time I began developing it as a kernel module.

People were happy about the results, and happy to contribute to the project.
This was the time when some users started reporting about others publishing the software on forums without consent. It felt like a bummer, I was not angry but kind of sad, especially because people were complaining on how it wasn't working properly and that their systems wouldn't work with the cameras.

Meanwhile, I started reverse engineering how the video was handled in the circular buffer. The kernel module idea was abandoned because of performance issues while interacting with userspace mapped memory.
Another re-write started and this time `viewd` was completely re-built from scratch, I used the FUSE library to create virtual devices and at this time I completely abandoned dvv's approach. This is the version that took longer to create, it included all of the streams available (HIRES, LOWRES and AAC). A lot of fixes to the streams' frames had to be made, some AAC headers were corrupted, some of the H264 NAL SPS units were incomplete and `viewd` had to fix all of this stuff to make the stream usable.
While working on this version, and seeing the issues about people re-distributing the software and complaining about it, I decided to create a simple DRM.

The logic was simple, a user would donate a small amount and they could **request as many licenses as they wanted**. 

This was perfect, people supporting the project could get RTSP on as many cameras as they had, or so I thought. 

The months passes and I started getting reports about people providing the licenses online. After some digging it turned out they were re-selling them and profiting from it.

It was frustrating. Imagine putting months of work into a project and see it being sold online. Random people profiting out of your work.

I was tired of people taking advantage of the situation and I made the decision to limit the license at one per donation. As you can see, this wasn't the direction I wanted to take but I didn't expect people acting like this.

Just a small note about the numbers: they aren't that high. A lot of donations were made when people could donate 50 cents (which would have turned out to be 0.00 because of PayPal fees).

I hope this makes things more clear.
If you have more questions let me know.

majkers, 2020-05-26:
Hey @TheCrypt0 I don't mind donating one more time (first was for RTSP) if You launch one more version with ONVIF for example. Great work and inspiration for many. Cheers!

kabturek, 2020-08-07:
You both have a valid point of view but i do think it should be clearly written in the main readme file that RTSP is a paid feature (minimum ~6.5$ per camera)

smashah, 2020-10-07:
I fully support your approach to licensing. I just think that automating the licensing procedure would've saved everyone, including yourself, a lot of headache.

fred-c1, 2021-01-22:
It should be clearly written in the main readme file that RTSP is a paid feature