TheCrypt0/yi-hack-v4 #201: 9FUS YI 1080p Home variant support?

punkonjunk, 2020-03-06:
Is this model in the near future for support?

pejotigrek, 2020-03-20:
+1 from me. just got the camera & looking for firmware. is there any roadmap on this?

benogen, 2020-03-20:
This should work for you, right: https://github.com/roleoroleo/yi-hack-MStar ?

pejotigrek, 2020-03-20:
...aaand it works great! thanks for the tip @benogen

nstrelow, 2020-04-05:
Is 9FUS using the MStar platform and needs https://github.com/roleoroleo/yi-hack-MStar ?

andyb2, 2020-04-14:
> Is 9FUS using the MStar platform and needs https://github.com/roleoroleo/yi-hack-MStar ?

There is a second version of 9FUS that is not MSTAR and it uses firmware 8.2.0.0A . There is no fix for it yet but it is being worked on by  roleoroleo. Could be a while. If your firmware is 4.5.0.0... then its probably MSTAR

pbanj, 2020-05-11:
https://github.com/roleoroleo/yi-hack-Allwinner just flashed this on mine, seems to be working well

nbendall34, 2020-10-14:
> https://github.com/roleoroleo/yi-hack-Allwinner just flashed this on mine, seems to be working well

How long did it take to flash Allwinner because mine has been running for 45mins now with a flashing blue light?

scruff-01, 2020-10-14:
What firmware is on the cam?

nbendall34, 2020-10-14:
> What firmware is on the cam?

8.2.0.0A_202007211802 currently

scruff-01, 2020-10-14:
Thats correct for allwinner. Make sure to format a low capacity sd card to fat 32. The sd cards have been temper mental for allwinner

nbendall34, 2020-10-14:
right, okay. I have another SD card on its way tomorrow anyway so I'll see how that goes. I took out the other card and rebooted the camera and it was fine. just weird I got stuck there...

scruff-01, 2020-10-14:
Yi cams are finicky with sd cards

nbendall34, 2020-10-14:
I might open a separate "issue" for this but I have just noticed that the app downloads firmware in an accessible format on Android phones, could it be swapped out with modified software then pushed to the device?

scruff-01, 2020-10-14:
You would have to ask on the allwinner github but I do not think so