TheCrypt0/yi-hack-v4 #204: Is there a way to turn on/off Yi Dome 1080 via web/ssh?

bit111, 2020-03-11:
Hi,
I've installed yi-hack-v4 on my Yi 1080 Dome camera and integrated it in home-assistant.
I would like know if there is a way to turn on/off camera outside yi app. My intention is to turn on/off it via home-assistant.

guillaume-smo, 2020-05-04:
any solution equivalent to the "on/off" button in Yi Home app ?

pkishino, 2020-05-05:
Kerois fork works well :)

mcgurdan, 2020-05-05:
How do I go about installing the Kerios fork?

pkishino, 2020-05-05:
Same as this one, go to fork and download latest release

bit111, 2020-05-05:
> edit: I found this fork https://github.com/Keroi/yi-hack-v4 which has implemented on/off switch "copied from official app", but I did not tested it

I have 2 cameras. I made a donation but I have enabled only one camera with RTSP. How can I enable second camera without make another donation?
Crypt0 project seems to be dead.

mcgurdan, 2020-05-05:
> Same as this one, go to fork and download latest release
> […](#)

The instructions on the fork to say to grab the firmware from the usual place:
https://github.com/TheCrypt0/yi-hack-v4/releases

However, the usual firmware does not include the extra functionality in that fork. Is there a guide on manually installing the files? The fork contains the yi-hack-v4-master folder?

bit111, 2020-05-05:
> 
> 
> > Same as this one, go to fork and download latest release
> > […](#)
> 
> The instructions on the fork to say to grab the firmware from the usual place:
> https://github.com/TheCrypt0/yi-hack-v4/releases
> 
> However, the usual firmware does not include the extra functionality in that fork. Is there a guide on manually installing the files? The fork contains the yi-hack-v4-master folder?

I founded this link: https://github.com/Keroi/yi-hack-v4/releases/tag/v0.2.1

mcgurdan, 2020-05-05:
That worked! Thanks

bit111, 2020-05-05:
> 
> 
> That worked! Thanks

Let me know if turn on/off works!

mcgurdan, 2020-05-05:
It does, I can see the web preview and everything seems to functioning properly (although RTSP stream is still very choppy).

However, when attempting to save the camera settings nothing happens. Looking at the dev tools in Chrome I can see a Javascript error:
_module camera_settings not registered_

tunerooster, 2020-08-19:
Using the 0.2.1 release, how do you turn the camera on/off using ssh or curl?

tunerooster, 2020-08-19:
I found the answer to my own question:

ssh camera ipc_cmd -t ON
ssh camera ipc_cmd -t OFF

Usage: ipc_cmd [t ON/OFF] [-s SENS] [-l LED] [-v WHEN] [-i IR] [-r ROTATE] [-m MOVE] [-p NUM] [-d]

	-t ON/OFF, --switch ON/OFF
		switch ON or OFF the cam
	-s SENS, --sensitivity SENS
		set sensitivity: LOW, MEDIUM or HIGH
	-l LED, --led LED
		set led: ON or OFF
	-v WHEN, --save WHEN
		set save mode: ALWAYS or DETECT
	-i IR, --ir IR
		set ir led: ON or OFF
	-r ROTATE, --rotate ROTATE
		set rotate: ON or OFF
	-m MOVE, --move MOVE
		send PTZ command: RIGHT, LEFT, DOWN, UP or STOP
	-p NUM, --preset NUM
		send PTZ go to preset command: NUM = [0..7]
	-f FILE, --file FILE
		read binary command from FILE
	-x,     --xxx
		send xxx message
	-d,     --debug
		enable debug
	-h,     --help
		print this help