TheCrypt0/yi-hack-v4 #205: New ip camera xiomi

ster1um, 2020-03-18:
Hello,
I see new camera

davimas, 2020-05-09:
Amy solutions? I have this MJSXJ02HL Camera here.

lucacalcaterra, 2020-09-04:
Follow...

av71966, 2020-09-16:
> Hey,
> 
> Any solutions?
> 
> I have this MJSXJ02HL Camera here.
> 
> Can we use this here?
> https://github.com/telmomarques/xiaomi-360-1080p-hacks

They have different CPUs.

davimas, 2020-09-16:
Yes - this hack in my Link dont work for this great Camera.

[image]

av71966, 2020-09-16:
> Yes - this hack in the link dont work for this great Camera :-(
> 
> [image]

Camera bsed on hi3518e rncv300 CPU like a lot of chinese ip cameras.

x2concepts, 2020-09-17:
follow...

TheBisk, 2020-09-25:
Hi,
have a look here:
https://xiaomifans.pl/topic/1066-mod-xiaomi-home-mi-home/?do=findComment&comment=29032

May this help us?
Thanks!

marines, 2020-10-05:
I've flashed the firmware and accessed telnet and ftp. There are rolling updates of video feed in `/tmp` directory. Like there are always 3 JPEG and 3 MP4 files containing the most recent snapshots with ever increasing number in the name (eg. 123.mp4, 124.mp4, 125.mp4). Clips are 3 secs long.

I don't really know which direction to go. I think I'll try to set up RTSP server somehow. While I'm fiddling with the camera maybe someone can use the information below. Let me know if you need anything. It's the magnetic MJSXJ02HL camera BTW.

```
/mnt/mmc # uname -a
Linux Hualai.fw 4.9.37 #7 Thu Oct 17 10:45:21 CST 2019 armv7l GNU/Linux
```

```
/mnt/mmc # mount
/dev/root on / type squashfs (ro,relatime)
proc on /proc type proc (rw,nosuid,nodev,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,relatime)
tmpfs on /tmp type tmpfs (rw,relatime)
tmpfs on /dev type tmpfs (rw,relatime)
tmpfs on /run type tmpfs (rw,relatime)
tmpfs on /mnt type tmpfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=600,ptmxmode=000)
/dev/mtdblock3 on /usr/app type squashfs (ro,relatime)
/dev/mtdblock4 on /backk type squashfs (ro,relatime)
/dev/mtdblock5 on /backa type squashfs (ro,relatime)
/dev/mtdblock6 on /configs type jffs2 (rw,relatime)
/dev/mmcblk0p1 on /mnt/mmc type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
```

```
/mnt/mmc # free
              total        used        free      shared  buff/cache   available
Mem:          40496       27808        1220        1180       11468       12056
Swap:             0           0           0
```

```
/mnt/mmc # df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                 3456      3456         0 100% /
tmpfs                    20248      1196     19052   6% /tmp
tmpfs                    20248         0     20248   0% /dev
tmpfs                    20248         4     20244   0% /run
tmpfs                    20248         0     20248   0% /mnt
/dev/mtdblock3            3712      3712         0 100% /usr/app
/dev/mtdblock4            1024      1024         0 100% /backk
/dev/mtdblock5            3712      3712         0 100% /backa
/dev/mtdblock6             384       240       144  63% /configs
/dev/mmcblk0p1        30234192   8218288  22015904  27% /mnt/mmc
```

```
/mnt/mmc # lsmod
Module                  Size  Used by    Tainted: P
hi_mipi_rx             32055  0
hi3516ev200_acodec      6820  0
hi3516ev200_adec        6039  0
hi3516ev200_aenc       29145  0
hi3516ev200_ao         58597  0
hi3516ev200_ai         60883  0
hi3516ev200_aio        26641  1 hi3516ev200_acodec
hi_sensor_i2c           1118  0
hi_pwm                  2090  2
hi3516ev200_ive       181357  0
hi3516ev200_jpege      44872  0
hi3516ev200_h265e     147193  0
hi3516ev200_h264e     122647  0
hi3516ev200_venc      273524  3 hi3516ev200_jpege,hi3516ev200_h265e,hi3516ev200_h264e
hi3516ev200_rc        100984  0
hi3516ev200_vedu       22193  0
hi3516ev200_chnl       33286  0
hi3516ev200_vpss      314506  0
hi3516ev200_isp       152450  0
hi3516ev200_vi        347815  0
hi3516ev200_vgs       131290  0
hi3516ev200_rgn        82291  0
hi3516ev200_sys       105497  0
hi3516ev200_base       56411 21 hi3516ev200_acodec,hi3516ev200_adec,hi3516ev200_aenc,hi3516ev200_ao,hi3516ev200_ai,hi3516ev200_aio,hi_sensor_i2c,hi3516ev200_ive,hi3516ev200_jpege,hi3516ev200_h265e,hi3516ev200_h264e,hi3516ev200_venc,hi3516ev200_rc,hi3516ev200_vedu,hi3516ev200_chnl,hi3516ev200_vpss,hi3516ev200_isp,hi3516ev200_vi,hi3516ev200_vgs,hi3516ev200_rgn,hi3516ev200_sys
hi_osal                56412 74 hi_mipi_rx,hi3516ev200_acodec,hi3516ev200_adec,hi3516ev200_aenc,hi3516ev200_ao,hi3516ev200_ai,hi3516ev200_aio,hi_sensor_i2c,hi_pwm,hi3516ev200_ive,hi3516ev200_jpege,hi3516ev200_h265e,hi3516ev200_h264e,hi3516ev200_venc,hi3516ev200_rc,hi3516ev200_vedu,hi3516ev200_chnl,hi3516ev200_vpss,hi3516ev200_isp,hi3516ev200_vi,hi3516ev200_vgs,hi3516ev200_rgn,hi3516ev200_sys,hi3516ev200_base
sys_config             10155  0
rtl8189ftv            785125  0
cfg80211              219243  1 rtl8189ftv
```

x2concepts, 2020-10-06:
Awesome, RTSP support would be great!

finnzz, 2020-10-27:
I'm thinking about getting this camera.  I see the WiF chip is dual band 2.4Ghz/5Ghz, but the product details only mention 2.4Ghz.  is it possible to enable the 5Ghz through a firmware change?

SIM0N-F, 2020-11-06:
Hey, I would to download this firemware for test it, but I cant download file on xiaomifan.pl, so have you copy of this files please ?

Thanks by advance!

fboulange, 2020-11-17:
> I've flashed the firmware and accessed telnet and ftp. 
> ```

The PL site is down unfortunatly
Any chance you kept thé firmware so we could have a go ? Looking forward to contribute enabling RTSP.
Cheers
Fred

albpara, 2020-11-17:
@marines @SIM0N-F for me is working, could you please try one more time? I have the files in any case

fboulange, 2020-11-18:
@albpara I would appreciate getting thé files
Can you share a link plz ? Cheers

al-one, 2020-11-18:
@fboulange @SIM0N-F 
https://dl.bintray.com/xiaomi/mjsxj02hl

fboulange, 2020-11-18:
@al-one Thank you !
Will investigate soon

SIM0N-F, 2020-11-19:
Yes many thanks @fboulange, with this file I can telnet on my camera and I have a busybox shell.

fboulange, 2020-11-19:
Cool, I managed as well to telnet and ftp my camera

I used the firmware_hlc6_4.0.5_0105_mod.zip content
Copy its content to SD card
Press RESET button with paper clip on camera while starting/plugin it

Now we have to find out how to adapt this PL custom firmware the same way @TheCrypt0  did with his "yi-hack-v4" to enable RSTP then some security (disable telnet and enable SSH).

Any tips are wellcome :)
Fred

PS I've got a strong background in Java developpement but a really weak on in Linux libraries....

karolzamosc, 2020-11-26:
Hi, any updates with RTSP on MJSXJ02HL? Regards

benleb, 2020-11-26:
RTSP works, tested a go rtsp server (because easy to compile) 👍

but there’s very low memory available and killing the xiaomi services the watchdog kicks in and reboots 🤪

so a very low memory rtsp server or disabling the watchdog and killing xiaomi stuff to free up some memory should work ✌️

karolzamosc, 2020-11-26:
Can you share the manual how to enable RTSP for a noob user? 🤪

benleb, 2020-11-26:
like I did already in my last comment?! but as said, you can’t use it currently, you can start it and seconds after watch ich to crash due to memory
or the watchdog

karolzamosc, 2020-11-26:
ok, I understand but did you try to disable the motion and person detection in Mi Home?

benleb, 2020-11-26:
nope, not checked... sounds as you also own such a cam? why don't you experiment with it yourself? :D

karolzamosc, 2020-11-26:
I have a lot of Xiaomi cameras (360 security, Imilab N2, Xiaovv D7, P1, Panoramic etc) and answer for your question is simple - I only use all cameras in Mi Home, some time ago I started play with Home Assistant but I am totally noob in programming and "hacking" cameras... :)

alnavasa, 2020-12-04:
Also interested! @benleb Thanks for the work so far.

paulomanuelp, 2020-12-10:
Hello,
What is the latest firmware link version that already supports the rtsp server for MJSXJ02HL?

fuzzznet, 2020-12-10:
> этому сервер rtsp с очень низкой памятью или отключение сторожевого пса и убийство вещей xiaomi для освобожд

Good afternoon! Can you post an instruction on how to start an RTSP stream?

marxo, 2020-12-10:
There are three open repos (in Chinese) that reference Hi3516ev200, although it isn't clear if this camera is using EV200 or EV300.

There is a kernel tree, a rootfs and uboot, all of which could be a great resource to try and build an open source firmware.
 https://github.com/topics/hi3516ev200

paulomanuelp, 2020-12-12:
Hello @benleb ,

Could you provide the rtsp source that do you use in your camera?

paunsurfer, 2020-12-28:
Any updates? I've been following progress on this. have a MJSXJ02HL which I would be willing to test / flash.

Revaes, 2021-01-29:
Any updates? I've bhave a MJSXJ02HL which I would also be willing to test / flash.

al-one, 2021-01-31:
https://github.com/al-one/hass-xiaomi-miot

This component can integrate xiaomi cameras to HomeAssistant.

fboulange, 2021-02-01:
In case you're looking for your tokens here is a tool : https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor

karolzamosc, 2021-02-01:
> https://github.com/al-one/hass-xiaomi-miot
> 
> This component can integrate xiaomi cameras to HomeAssistant.

Thanks for reply but it does not work for me. I installed repisotory in HACS, then I added MJSXJ02HL camera in Configuration > Integration > ➕ > Xiaomi Miot providing IP and token but nothing happened then... :(

TheBisk, 2021-02-02:
[image]

I see the camera entity but with no image stream.

al-one, 2021-02-02:
@karolzamosc @TheBisk  Do you enabled the [miot_cloud](https://github.com/al-one/hass-xiaomi-miot#configuration-xiaomi-cloud) ?

fboulange, 2021-02-03:
At least I got a feed in Home Assistant using @al-one  (thanks to him) with this MJSXJ02HL (isa-camera-hlc6)

For detail in : https://github.com/al-one/hass-xiaomi-miot/issues/11#issuecomment-772371095

bahbka, 2021-02-12:
@al-one Can you help me please to get feed in HA?
At first after adding integration it was exception and no devices was added:
```
2021-02-12 17:31:01 ERROR (MainThread) [homeassistant.components.camera] Error while setting up xiaomi_miot platform for camera
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 198, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/xiaomi_miot/camera.py", line 45, in async_setup_entry
    await async_setup_platform(hass, config, async_add_entities)
  File "/config/custom_components/xiaomi_miot/camera.py", line 64, in async_setup_platform
    entities.append(MiotCameraEntity(hass, cfg, srv))
  File "/config/custom_components/xiaomi_miot/camera.py", line 85, in __init__
    super().__init__(name, self._device, miot_service)
  File "/config/custom_components/xiaomi_miot/__init__.py", line 595, in __init__
    super().__init__(name, device, miot_service, **kwargs)
  File "/config/custom_components/xiaomi_miot/__init__.py", line 397, in __init__
    super().__init__(name, device, **kwargs)
  File "/config/custom_components/xiaomi_miot/__init__.py", line 277, in __init__
    'hardware_version': self._miio_info.hardware_version,
  File "/usr/local/lib/python3.8/site-packages/miio/device.py", line 85, in hardware_version
    if self.data["hw_ver"] is not None:
KeyError: 'hw_ver'
```

I commented line 277 at xiaomi_miot/__init__.py `'hardware_version': self._miio_info.hardware_version,` and devices was 
added but no feed available. Now I see new exception:
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 278, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 474, in async_device_update
    raise exc
  File "/config/custom_components/xiaomi_miot/camera.py", line 132, in async_update
    await super().async_update()
  File "/config/custom_components/xiaomi_miot/__init__.py", line 463, in async_update
    attrs = {
TypeError: 'NoneType' object is not iterable
```

And when trying to open live stream `Error requesting stream: Camera is off`.

I've enabled miot_cloud, defined miot_did without success :( My region is ru.

al-one, 2021-02-19:
@bahbka  Maybe you should use `miiocli` to your camera.

```shell
pip3 install python-miio
miiocli -d device --ip 192.168.lan.ip --token xxxxxx info
```

Capsman4, 2021-03-20:
Hi , 
Little question for a french newbee  : i need the custom firmware in my camera , it's Ok , but i don't know how to connect to ssh on my camera or connect to http web. 

I test modify run.sh with ip address camera but putty ssh nok and http://ipcam not working. 
Please to help me
thank a lot

bahbka, 2021-03-26:
> @bahbka Maybe you should use `miiocli` to your camera.
> 
> ```shell
> pip3 install python-miio
> miiocli -d device --ip 192.168.lan.ip --token xxxxxx info
> ```

Same error, `KeyError: 'hw_ver'` :( also after fixing this error I tried to run command `get_ipcprop ['all']`, it responds with ['OK'] without any useful info. It seems there is no way to get stream or image from my cam MJSXJ02HL :(

john9393, 2021-03-31:
@ Capsman4 
with the custom poslish firm  there is NO SSH or HTTP server
you can  use only TELNET or FTP  (putty can be used to connect on TELENt server)

Capsman4, 2021-04-01:
@ jonh9393 , thank you , I think I have a problem, I connect with putty in telnet but it does not work.

Is there a login and password required when I connect to the camera?

john9393, 2021-04-02:
Telnet / FTP User root Pass ismart12

the SD card with the special firm must stay in the camera

grifu, 2021-05-02:
I'm trying to configure this camera with Home Assistant and it is required Telnet access, and MIOT integration.
My problem is placing the modified version of the firmware (firmware_hlc6_4.0.5_0105_mod) into to the camera. 
The camera accepts the original version which is signed (demo_hlc6_sign.bin). However, if I try to use the Modified version which comes with just one file (demo_hlc6.bin) or even the original version (without signed file) it simply does not work. The camera just resets the configuration and hear the voice for the QR code. I've tried combining the two files one from the Mod version (demo_hlc6.bin) and the other from the original version (demo_hlc6_sign.bin)  and placing the run.sh in the root of the card. The camera downgrades the firmware but there is no telnet!
How can I check if the version installed is the modified one without trying to "telnet IP" to the camera?
Anybody found this issue when trying to install the modified version?

borewik, 2021-05-05:
> [...]
> How can I check if the version installed is the modified one without trying to "telnet IP" to the camera?
> Anybody found this issue when trying to install the modified version?

You need to connect the camera with your wifi network via xiaomi app. Then you will know the ip address. Once you got it do as instructed, put bin and run.sh on sd card and turn it while keeping reset button pressed. After that you can telnet camera via ip

grifu, 2021-05-06:
I've already done this procedure. However, the camera downgrade the firmware only with the signed version "demo_hlc6_sign.bin". Reseting the camera with this file in the sdcard the LED goes white and the firmware is upgraded. With the modified version or the unsigned version "demo_hlc6.bin" in the sd card the camera just resets the settings and hear the voice to start the configuration. Cannot access to Telnet even if the "run.sh" script is in the sdcard. Strange behaviour!

borewik, 2021-05-06:
> I've already done this procedure. However, the camera downgrade the firmware only with the signed version "demo_hlc6_sign.bin". Reseting the camera with this file in the sdcard the LED goes white [...]

Modification that runs sh script from mmc card was previously done in app.bin/bin/init_app.sh, by the looks of it those files are exactly the same and are not modified in both firmware bins from "firmware_hlc6_4.0.5_0105_signed.zip". Maybe that change was made somewhere else, didn't check.
[Try this one](https://mega.nz/file/lsxEBKiL#RoE2_7oOu6mNgyN2a-WgqiOj1lV6vgDYEnKYoIYgLQ8), it is signed version with same modification as was made in previous telnet working fws.

[image]

grifu, 2021-05-06:
Thank you for sharing the file. I've tried to install the firmware without luck. No White light, just voice "Reset successful ...". There is a diference in size between the files, the signed version that works for me "demo_hlc6_sign.bin" is 11,1 MB and all the other versions that I've tried including yours "demo_hlc6.bin" are 12,1 MB. I'm copying the files to the card in OSX. I can't understand this behaviour.

grifu, 2021-05-09:
The file that works for me can be found in this [link](https://mega.nz/file/4aoEhBga#wUr1gcrM3rNU_Jz-QYirdHNZgZGXTcFQI0nOWLitdf8). It was retrieved from this [site](https://npirtube.com/foro/viewtopic.php?t=12220).
There is another file with the same firmware in the same site that does not work, you can find in this [link](https://1drv.ms/u/s!Ar5C-aX9A4YA0W8gjUuhi6YctCc3). The working version 4.0.5_0105, provides to files (demo_hlc6_sign.bin and demo_hlc6.bin) and seems to be un update from the older version 4.0.5_0105 which provides just one file demo_hlc6.bin. However, this file does not work for me. The size of this older version (4.0.5_0105) is 12,1 Mbytes and the the newer version and signed version are 11,1 Mbytes. Why the differences in these files for the same version?

tomaszduda23, 2021-05-19:
@grifu did you manage to run telnet or ftp?

tomaszduda23, 2021-05-22:
Yes. Signed firmware does not have run.sh

borewik, 2021-05-22:
I made a mistake by not checking how this signed fw structure look like I just used same "packer.py" as for previous fws. Gotta run binwalk on that signed fw, and make changes to packer.py in line 16 "dic" put there correct addresses for partitions. Can do that correctly for you this time, if that info doesn't help you enough.

tomaszduda23, 2021-05-22:
Modification is easy but I though that it will be rejected by bootloader due to incorrect signature. I have not verified it though. I will check. I will be surprised if it works.

$ hexdump demo_hlc6_sign.bin | tail
0a8d930 0000 0000 0000 6000 0000 0000 0000 bc00
0a8d940 0000 0000 0000 ed00 0fd8 0000 0000 0800
0a8d950 e880 0003 0000 0000 0f00 0fd9 0000 0000
0a8d960 0000 0000 0000 0000 0000 0000 0000 0000
*
0a8e040 0484 96c0 4570 a32d 7856 69c5 cab5 4f1a  <--- it seems to be signature
0a8e050 8161 e90a c23b a26f 75bf 6054 035a e275
0a8e060 9546 9e74 984d ab04 9aad 3e28 7731 4ee9
0a8e070 6007 4d90 fa5e 5831 19be aa35 c776 0634
0a8e080

$ hexdump demo_hlc6.bin | tail
0a8d950 e880 0003 0000 0000 0f00 0fd9 0000 0000
0a8d960 0000 0000 0000 0000 0000 0000 0000 0000
*
0a8e040 0484 96c0 4570 a32d 7856 69c5 cab5 4f1a
0a8e050 8161 e90a c23b a26f 75bf 6054 035a e275
0a8e060 9546 9e74 984d ab04 9aad 3e28 7731 4ee9
0a8e070 6007 4d90 fa5e 5831 19be aa35 c776 0634
0a8e080 0000 0000 0000 0000 0000 0000 0000 0000 <--- unsigned version is padded with zeros
*
0b7f080

grifu, 2021-05-22:
In did, there are differences in the unsigned version which is padded with zeros as tomaszduda explains. Is it possible to sign the firmware to trick the boot loader?

rezmus, 2021-06-16:
it's possible to flash old bootloader which accepts unsiged firmware. here you can find all info and custom firmware with rtsp.

https://io-net.ru/proshivka-zagruzchika-ip-kamery-mjsxj02hl-s-pomoshhyu-usb/
https://io-net.ru/mjsxj02hl_firmware/

grifu, 2021-06-17:
In did Rezmus, it is possible to flash the boot loader, but it requires opening the camera and soldering some wires which I would prefer to avoid. Is it possible to check which version of the boot loader is installed in our cameras?

rezmus, 2021-06-17:
check links i gave. it's possible to flash bootloader with hitool without opening the camera.

grifu, 2021-06-17:
Tnx Rezmus, I did knew! it would be great to change the boot loader without messing with the hardware. I'll check the links.

[12 more replies]