TheCrypt0/yi-hack-v4 #206: Rotation Control

metalnodeug, 2020-03-18:
Hi

Have you an idea to control the rotation on this yi cam maybe with Synology Live View ?

Because when I reboot the cam, she reset the position and because I deactivate yi cloud, I can only manually rotate the cam.

This is my dream :P Can you help me please ?

GauntPT, 2020-04-13:
Hi metalnodeug,

I have raised the same question how to manually move the camera:
https://github.com/TheCrypt0/yi-hack-v4/issues/27
Unfortunately no answer till today. 

What I know is that the rotations are known commands. You can have a look here:
https://github.com/roleoroleo/yi-hack-MStar/issues/44
But I never figured out how to send them. Maybe you are more skillfull than me.

Crypto was also working on it and he wrote in an earlier thread that it is working. But since then this project is unfortuantely kind of dead :'(

Good luck!

metalnodeug, 2020-04-13:
> Hi metalnodeug,
> 
> I have raised the same question how to manually move the camera:
> #27
> Unfortunately no answer till today.
> 
> What I know is that the rotations are known commands. You can have a look here:
> [roleoroleo/yi-hack-MStar#44](https://github.com/roleoroleo/yi-hack-MStar/issues/44)
> But I never figured out how to send them. Maybe you are more skillfull than me.
> 
> Crypto was also working on it and he wrote in an earlier thread that it is working. But since then this project is unfortuantely kind of dead :'(
> 
> Good luck!

Thanks a lot GauntPT, I contact roleoroleo for this. 
Unfortunately you are right, this project is dead for this time.