TheCrypt0/yi-hack-v4 #212: Unable to donate - Paypal issue

mcgurdan, 2020-03-29:
I am not the only person reporting this in the Discord channel, but there seems to be an issue when attempting to donate via Paypal.

Thanks

[image]

TheCrypt0, 2020-03-31:
Hi @mcgurdan 
Thank you for the report.

I'm having some issues with PayPal too (especially that generic "Sorry, there's a problem." error). I usually empty Chrome's cache, close and re-open it and then it works.

Sometimes even this procedure doesn't work, in that case you can try to firstly log-in into your PayPal account (bypassing the "One Touch" login) and then try again.

Let me know if it helps!

kirilldem, 2020-04-04:
> Hi @mcgurdan
> Thank you for the report.
>
> I'm having some issues with PayPal too (especially that generic "Sorry, there's a problem." error). I usually empty Chrome's cache, close and re-open it and [...]

Do you have plans to resume yihackv4 development?

tnr20, 2020-12-14:
Same here... i wait now over 9months . i said him that i payed already 9months ago and he banned me from discord

TheCrypt0, 2020-12-15:
@tnr20 I've replied to you here: https://github.com/TheCrypt0/yi-hack-v4/issues/190#issuecomment-745337321