TheCrypt0/yi-hack-v4 #213: Not working with yi 1080p (4.5.0.0C_201910080934)

fjcabello, 2020-03-29:
Hi,
with a 32GB SDHC and model yi home 1080p with firmware version 4.5.0.0C_201910080934 the firmware is not loaded. 
I follow these steps:
1. Switch off camera
2. Put SD Card with files  home_y20 and rootfs_y20.  SD Card is formatted with FAT32 and only these files in the root file system.
3. I switch on the camera

After these steps the camera boots normally without installing new firmware.

Any ideas?

Thanks

rabbitsf, 2020-03-29:
Same problem here. I have have the 1080p Yi Dome with firmware version 4.6.0.0A_201908271549. 
I followed the instructions. But the cam just boots normally without installing anything.

rabbitsf, 2020-03-29:
> Hi,
> with a 32GB SDHC and model yi home 1080p with firmware version 4.5.0.0C_201910080934 the firmware is not loaded.
> I follow these steps:
>
> 1. Switch off camera
> 2. Put SD Card with files  home_y20 [...]

I found this instead:
https://github.com/roleoroleo/yi-hack-MStar

fjcabello, 2020-03-30:
Thanks! I managed to install the firmware using the yi-hack-MStar.

closed 2020-03-30 by fjcabello