TheCrypt0/yi-hack-v4 #214: Live feed

ebolisa, 2020-03-30:
Hi, 
I installed the software and changed some settings. So, I can connect via SSH and browser.
The question I've is, how can I see live feed from the camera? The snapshot works but I'm mostly interested in the video.
I tried rtsp://my_ip/ch0_0.h264 with no avail.
What's the trick if any?
TIA

Mcgurk125, 2020-03-30:
You need to follow the guide on enabling RTSP

ebolisa, 2020-03-30:
Hi, It’s already enabled.

Mcgurk125, 2020-03-30:
Use an RTSP app, like VLC media player, open network stream and put in your RTSP url

Mcgurk125, 2020-03-30:
https://www.unifore.net/ip-video-surveillance/how-to-play-rtsp-video-stream-of-ip-cameras-on-vlc-player-quicktime-player.html

ebolisa, 2020-03-30:
I did also try VLC. Will follow the link. Thanks

Mcgurk125, 2020-03-31:
Did you get it working?

ebolisa, 2020-03-31:
Still at it. I'm in quarantine (day 17th) and every so often internet drops out. Will report as soon as possible. Thank you for asking.

ebolisa, 2020-03-31:
Ok, got as far as introducing my user/pass credentials and then I get theses errors:

Your entry cannot be opened:
VLC is unable to open the MRL <<rtsp://192.168.0.41:554/h264/quad/media.amp>>
VLC is unable to open the MRL <<rtsp://192.168.0.41:554/mjpeg/quad/media.amp

I also tried MQTT so send a motion detect signal to my mosquitto server with no avail but, as I said earlier, I've no problem SSH into the device.