TheCrypt0/yi-hack-v4 #221: Wifi connection and camera initialization 4FUSY21N44

alexkutsan, 2020-04-11:
Hi, I used yi-hack-v4 for a while, but after some time camera turned on and with blinking yellow and "Waiting to connect" prompt.
And now I am not able to add camera to the app.
Procedure stops after scanning QR code. 
Camera says that wifi connection successful, but app is not able to fund it. And after some time camera says that "Connection timeout, please reboot the camera".
And after rebooting "Waiting to connect" again. 
What can i do?

alexkutsan, 2020-04-11:
After some investigation found out multiple error resolving host names.
Looks like DNS configuration failed. 
```/tmp # cat log.txt | grep resol
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
][1/19/6:48:0:806]: p2p_tnp.c(tnp_proc-6408) P2P Server IP resolved : NO
][1/19/6:53:20:807]: p2p_tnp.c(tnp_proc-6408) P2P Server IP resolved : NO

```

how can I fix dns issues?

jurgen272, 2020-04-16:
Hi, same issue here. Cam connects to network and is reachable through the webpage, but adding it to the app fails

jurgen272, 2020-04-16:
Activated cloud and rebooted and is working again now.

alexkutsan, 2020-04-23:
`Activated cloud`, what actualy do you mean?

jurgen272, 2020-04-23:
On the webpage of the device (Enter ip-address in browser) you can activate en desactivate cloud connection. It was not working with cloud disabled. I Enabled the cloud and could ad the camera to the Yi app. After that you can desactivate cloud again I think.

StreneSaco, 2020-06-21:
Nope, same problem here. Enabling and disabling cloud connection doesn't change anything. It happened after a change of wifi router, so i had to reinitialise the webcam in order to connect to the new wifi

alexkutsan, 2020-06-21:
Enabling and disabling cloud connection doesn't change anything for me to

Johnnyyr, 2020-10-28:
same problem here, what can we do? another dns?