TheCrypt0/yi-hack-v4 #222: Loop recording on SD Card YI Outdoor

Casalex97, 2020-04-12:
I have yi hack on Yi outdoor 1080 is possible loop recording on 32 GB SD Card on board?

Thank you

Limag, 2020-05-06:
Yes, the cam creates 1 minute *.mp4 videos, put in directories, each for 1 hour.
I have the feeling that the recording stops if the sd card is full. So I setup a cron job on the camera.
```
crontab -e
 /usr/bin/find /tmp/sd/record/* -mtime +5 -delete

```
If you setup your camera with the app, check this settings:

Settings
    ->MicroSDCard: In good condition (or format...)
    ->Activity Detection Recording: OFF (record always)

Casalex97, 2020-05-08:
> Yes, the cam creates 1 minute *.mp4 videos, put in directories, each for 1 hour.
> I have the feeling that the recording stops if the sd card is full. So I setup a cron job on the camera.
>
> ```
> crontab [...]

Ok thank you, i use this camera with rtsp server to a synology NAS. If the wifi network for some reason fail how can i watch this recorded videos in SD? I have to extract the SD card and view it on a computer or it's possible to copy later the file on NAS?
Which app are compatible with Yi-Hack?

Thank you

photonforge, 2020-05-14:
I use a shell script to copy files from the sd card via ftp to a HDD attached to a raspberry pi

> #!/bin/bash
HOST='10.0.1.26:21'
USER='******'
PASS='******'
TARGETFOLDER='/mnt/storage/cctv/yicamftp/'
SOURCEFOLDER='/tmp/sd/record/'
lftp -f "
open $HOST
user $USER $PASS
lcd $SOURCEFOLDER
mirror -c --ignore-time --no-empty-dirs --use-cache $SOURCEFOLDER $TARGETFOLDER
bye
"

I also discovered that if we move ALL the files from the /tmp/sd/record folder, the camera stops recording further, so i created another script to move only files more that 2 days old, making sure some files are left behind on the camera

> lftp -f "
open $HOST
user $USER $PASS
lcd $SOURCEFOLDER
mirror -c --ignore-time --no-empty-dirs --use-cache $SOURCEFOLDER $TARGETFOLDER
bye
"

I then created a cron job to run both these scripts every 1 hour. So the first script copies all videos to the HDD and the second script deletes anything more than 2 days old form the camera. 

This has been working flawlessly on my yi home 720p.
Much more reliable than trying to record from an RTSP stream using motioneye etc..

Limag, 2020-05-14:
I have a different approach: If I get a move event via MQTT. Then, I copy the affected files with scp and create some JPEGs right now from the motion-start position for fast examination.

Here are the example commands, login via private key. I tried also to optimize the SSL encryption for this tiny cam hardware but the speed up was marginaly, anyway.

From the host computer:
```
$ scp -c aes128-cbc -i ./cams root@poopy:/tmp/sd/record/2020Y05M04D07H/25M00S.mp4 work.mp4
$ ffmpeg -v 0 -ss 18.0 -i work.mp4 -t 3 -r 1 -s 480x300 -f image2 img_2020Y05M04D07H25M00S_%0d.jpg
```

1onar, 2020-05-27:
> Yes, the cam creates 1 minute *.mp4 videos, put in directories, each for 1 hour.
> I have the feeling that the recording stops if the sd card is full. So I setup a cron job on the camera.
>
> ```
> crontab [...]

Sorry to ask but i am a noob. Where do i insert this crontab script?
And what would be the command if i want to delete all records that are 48 hours and older?
For example i want to delete every 2 days the Recordings that are 48+ hours old.

pkishino, 2020-05-27:
Setup the crontab on the camera..
And adjust the mtime to 48 hours.. google what value is needed for this

1onar, 2020-05-27:
> Setup the crontab on the camera..
> And adjust the mtime to 48 hours.. google what value is needed for this

Well yeah. But where do set the crontab?

pkishino, 2020-05-27:
ssh into the camera and run crontab -e

Limag, 2020-05-27:
On the camera runs a Linux, so any Linux tutorial is useful. Just google for the commands **contab** and  **find**. Man pages or howtos, what you like. Don't care about Linux distributions (RedHeat, Debian etc), the commands are always the same. 
Small hint: Its a very tiny Linux on your camera, maybe the command does not support all options related to a desktop linux. **Busybox** is a good keyword to google for embedded Linux with limited functionality.

1onar, 2020-05-27:
When i do crontab -e i cannot edit the file. What do i need to enter before

/usr/bin/find /tmp/sd/record/* -mtime +5 -delete

Limag, 2020-05-27:
Google for "vi" :-)

1onar, 2020-05-27:
I edited the file "root" in "/home/yi-hack-v4/etc/crontabs" but it doesn't change anything. Shouldnt it delete all records that are 5min and older?

[image]

pkishino, 2020-05-27:
I sshd into the camera, then ran "crontab -e"
it opens the file, can enter a new line
then esc,type :wq
and enter
done
check crontab -l for list of crontabs installed

1onar, 2020-05-27:
> I sshd into the camera, then ran "crontab -e" it opens the file, can enter a new line then esc,type :wq and enter done check crontab -l for list of crontabs installed
> […](#)

Something isnt working as intended for me then. I cant edit the crontab when i use ssh.

While i browsed via ftp through the files there were some temporarily crontab files like "crontab462619" and "crontab1733" in there. There is always one more if try to edit the file via ssh as root.

I edited the crontab file via ftp like the Screenshots aboth. I doesnt work..

pkishino, 2020-05-28:
what do you mean with you cannot edit the crontab?
can you run crontab -e? errors?

1onar, 2020-05-28:
I cant edit it via SSH. It creates a new Crontab File and i cant save it.
[image]

I downloaded the file "root" via ftp, edited it and saved it to "/home/yi-hack-v4/etc/crontabs". It seems to be there but doesnt have a effect on the recorded videos.

[image]
[image]
[image]

pkishino, 2020-05-28:
Yes, well, you haven’t given it a schedule.. see the first command and
google how to create a cron  schedule .
Right now it’s missing

1onar, 2020-05-29:
> Yes, well, you haven’t given it a schedule.. see the first command and google how to create a cron schedule . Right now it’s missing
> […](#)

Sorry to annoy you. But it's still not working.

[image]

I did it with a Crontab Generator.
`0 11 * * * /usr/bin/find /tmp/sd/record/* -mtime +1440 -delete >/dev/null 2>&1`

It should delete everything that is older than 24hours everyday at 11 UTC.

pkishino, 2020-05-29:
ok, try trigger
/usr/bin/find /tmp/sd/record/* -mtime +1440 -delete
manually and see if it works

1onar, 2020-05-29:
> ok, try trigger /usr/bin/find /tmp/sd/record/* -mtime +1440 -delete manually and see if it works
> […](#)

I dit it right now. No error but the records 24+ hours are still there.

[image]

pkishino, 2020-05-29:
does /usr/bin/find /tmp/sd/record/* -mtime +1440 show anything?
most likely your dates or value are incorrect

1onar, 2020-05-29:
No, nothing happens when i enter it.

pkishino, 2020-05-29:
are there any records in /tmp/sd/record/ ??

Limag, 2020-05-29:
Could you follow the instructions https://github.com/TheCrypt0/yi-hack-v4/issues/222#issuecomment-634538212 carefully, please?

If you did, you would read in the man page 

> 
> -mtime n
>     File's data was last modified n*24 hours ago. See the comments for -atime to understand how rounding affects the interpretation of file modification times. 

If you don't get it, read it again until you recognize that your variable in **n**. And **n** is a day!
Please, don't just copy&paste from the internet, it will never work. Yes, copy from the internet is good, I call it learning. But the next step is not paste, the next step must be **understand** whats going on. 

Tip: The plus sign before n is important, it means:
```
    +n More than n.
      n Exactly n.
    -n Less than n
```

1onar, 2020-05-29:
> Could you follow the instructions [#222 (comment)](https://github.com/TheCrypt0/yi-hack-v4/issues/222#issuecomment-634538212) carefully, please?
> 
> If you did, you would read in the man page
> 
> > -mtime n
> > File's data was last modified n*24 hours ago. See the comments for -atime to understand how rounding affects the interpretation of file modification times.
> 
> If you don't get it, read it again until you recognize that your variable in **n**. And **n** is a day!
> Please, don't just copy&paste from the internet, it will never work. Yes, copy from the internet is good, I call it learning. But the next step is not paste, the next step must be **understand** whats going on.
> 
> Tip: The plus sign before n is important, it means:
> 
> ```
>     +n More than n.
>       n Exactly n.
>     -n Less than n
> ```

Thank you. I know you want to teach me. I ran the command
`/usr/bin/find "/tmp/sd/record/" -mtime +1 -delete`
and i deleted everything 1+ hour ago.

1onar, 2020-05-30:
I don't get it to working. I have tried those commands to test it:

`/usr/bin/find "/tmp/sd/record/" -name "*.mp4" -mtime +1 -delete`
`find /tmp/sd/record -type f -name '*.mp4' -mtime +1 -delete`
`find /tmp/sd/record -mtime +1 -delete`
`/usr/bin/find /tmp/sd/record/* -mtime +1 -delete`

Can anybody help me please.

pkishino, 2020-05-30:
Look, if you want help then at least provide more information.. what is not
working, what happens and what do you expect, any errors and logs etc?
This is NOT a Linux support forum...

1onar, 2020-05-30:
> Look, if you want help then at least provide more information.. what is not working, what happens and what do you expect, any errors and logs etc? This is NOT a Linux support forum...
> […](#)

`/usr/bin/find /tmp/sd/record/* -mtime +1 -delete`
no error; system needs 1-2 sec to "load" (all records are still there).

`/usr/bin/find "/tmp/sd/record/*" -mtime +1 -delete`
find: /tmp/sd/record/*: No such file or directory

`/usr/bin/find /tmp/sd/record/ -mtime +1 -delete`
no error; system needs 1-2 sec to "load" (all records are still there).

`/usr/bin/find "/tmp/sd/record/" -name "*.mp4" -mtime +1 -delete`
no error; system needs 1-2 sec to "load" (all records are still there).

`find /tmp/sd/record -type f -name '*.mp4' -mtime +1 -delete`
no error; system needs 1-2 sec to "load" (all records are still there).

`find /tmp/sd/record -mtime +1 -delete`
no error; system needs 1-2 sec to "load" (all records are still there).

1onar, 2020-05-31:
If i do `ls -al /tmp/sd/record/*` all files are there with the right dates.

[image]

But none of the commands i am trying to use are working. I want to delete every record that is older then 1 day.

`find /tmp/sd/record/* -name '*.mp4' -mtime +1 -delete`
`/usr/bin/find /tmp/sd/record/* -name "*.mp4" -mtime +1 delete`
`/usr/bin/find /tmp/sd/record/* -mtime +1 delete`
`find /tmp/sd/record/* -type f -mtime +1 -exec rm -f {} \;`
`find /tmp/sd/record/ -type f -mtime +1 -exec rm -f {} \;`
`find /tmp/sd/record/ -type f -mtime +1 -exec rm -f {} \;`

Limag, 2020-05-31:
Testing your command interactive, without delete option lists the affected files.

```
/home/yi-hack-v4 # /usr/bin/find /tmp/sd/record/* -mtime +1
/tmp/sd/record/2020Y05M27D13H
/tmp/sd/record/2020Y05M27D13H/52M00S.mp4
/tmp/sd/record/2020Y05M27D13H/53M00S.mp4
<and more...>
/tmp/sd/record/2020Y05M29D11H/55M00S.mp4
```
With mtime +1, the newest affected file is ~48h old, how can it be? If you would start reading the man page, you would figure out:
_-atime n
    File was last accessed n*24 hours ago. When find figures out how many 24-hour periods ago the file was last accessed, any fractional part is ignored, **so to match -atime +1, a file has to have been accessed at least two days ago.**_ 

And I agree: "This is NOT a Linux support forum...". Please ask additional questions, not related to the camera firmware in a linux support forum. Please read man pages and real documentations before post anything!

For completeness, this is my crontab, and it works!
```
/home/yi-hack-v4 # crontab -l
#12 2 * * * /home/yi-hack-v4/script/check_update.sh
#delete files older +n+1  days
12 2 * * * /usr/bin/find /tmp/sd/record/* -mtime +3 -delete
/home/yi-hack-v4 # 

```

x3hong, 2020-10-12:
manul excute crond  then it will be ok

eilz, 2021-11-16:
Would really appreciate some assistance on below:  (I have a outdoor YI using all-winner FW)

note : I don;t have the "crontab" command/file in the system so can run this

If I do a **ps** I see this line:
**1408 root       864 S    /tmp/sd/yi-hack/usr/sbin/crond -c /var/spool/cron/crontabs/**

I also created a the file 
# cat /tmp/sd/yi-hack/script/**del_morethan4days.sh**
**/tmp/sd/yi-hack/usr/bin/find /tmp/sd/record/* -mtime +3 -delete**

I then edit below 
vi /tmp/sd/yi-hack/script/**system.sh**

and added this below line
**echo "12 2 * * * /tmp/sd/yi-hack/script/del_morethan4days.sh" >> /var/spool/cron/crontabs/root**

Can anyone help me figure out how I get it to work, if I run the **del_morethan4days.sh** manually it works ok.