TheCrypt0/yi-hack-v4 #224: is there an easy way to reset password?

pejotigrek, 2020-04-16:
ok, so I was stupid enough, not to add the password to my KeePass database, and after flashing/configuring camera weeks ago, I left it in the drawer to wait for the day I'll be installing it in renovated room. today is that day and guess what - I can't login, because I don't remember the password... camera connects to my wifi, so it is configured correctly.

can I reset the password without reflashing? or even reflash won't help?

TheCrypt0, 2020-04-16:
Hi,
Re-flashing will help, to do so you will have to download an older version
of the firmware, flash it and then re-flash the latest version. This is
needed because the camera detects if you are trying to flash the same
version and prevents it.

Another way could be to attach via UART to the camera and (if the autologin
is enabled) change the password from there. But I'm sure it's easier to
re-flash ;)

pejotigrek, 2020-04-16:
so reflash it is :)

thank you for such a quick response!

closed 2020-04-16 by pejotigrek