TheCrypt0/yi-hack-v4 #225: Yi Dome camera 1080p

Dani98b, 2020-04-16:
good evening everyone, I would like to know how I can, access the web interface on my Yi dome camera1080p I wrote the firmware put it inside the sd card but it doesn't blink me in yellow and then it returns normal to me because I don't know how to do any of you know how can I do I would like to access on rtsp, and many, others. Contents who can help me please?

subasically, 2020-05-11:
I am having the same issue...

subasically, 2020-05-11:
> I am having the same issue...

So I got mine working. I copied the files to the microsd card and the reset the camera. I then went through the normal setup by hooking the camera up to my wifi and then once that was done I was apple to access the camera but navigating to the cameras ip address.

Dani98b, 2020-05-11:
Yes this Problem is too strog im a yi 1080p WiFi ot rtcp im to workig for my camera whts the Problem?