TheCrypt0/yi-hack-v4 #226: Yi Outdoor Camera XY-R9520-V3

larsoss, 2020-04-20:
Hello,

It doesn't matter what I try, but if I format my sd card to Fat32, and put the files in the root, then turn the camera on (plug in socket), nothing else happens than the camera starts up.

Adjrobinson, 2020-04-30:
looks like there are a few of these floating around on eBay for £30 ish, they appear to be the Chinese version and I can't pair it with the international app, tried a few different firmware now with no luck, don't fancy getting the Chinese app, is anyone able to help?

Adjrobinson, 2020-04-30:
I just opened it up and it appears that there is a 4 pin connector on it, might be able to putty into it, anyone willing to help? Just need to see if I can get some jumpers on to it or find a connector etc? Might take me a couple of days to do that due to new baby but it would be nice to get it working.
[image]

rage2dev, 2020-05-09:
I went down the same road. On the new versions of yicam, they disabled telnet and tty/uart. 

Solution:
https://github.com/rage2dev/YiOutdoor

But we need the RT-Sdk to compile rtsp_server :/

larsoss, 2020-05-09:
So whe need to wait a little bit more.

JhonnyMarr, 2020-05-10:
I´m also interested in yi_Hack_v4 for the Yi IOT outdoor camera (model XY-R9520-V3). Very similar to yi outdoor https://es.aliexpress.com/item/4000120325499.html?spm=a2g0s.9042311.0.0.274263c06jCm4M
I´ve got one of them. Firmware version 7.1.00 11A201905311440  I´ve seen that the telnet port is open (user: root no passwd)
If any coder needs something tell me what to do.
It has the option to stream to an Alexa Echo Show; Would it be possible to capture that stream with VLC or similar?
It has a Mobile and PC app http://www.yunyiiot.com/html/download_en.html
Regards

DrumSergio, 2020-05-22:
> I went down the same road. On the new versions of yicam, they disabled telnet and tty/uart.
> 
> Solution:
> https://github.com/rage2dev/YiOutdoor
> 
> But we need the RT-Sdk to compile rtsp_server :/

Great contribution, thank you!
But that means there's no actual way to flash YiHackV4 or YiHackMStar, right?
I acquired this Yi IoT Outdoor 1080p cam thinking it was going to be easy - seems not.

Abuelika, 2020-05-22:
I am in the same situation. I have several cameras and at the moment in the box

> > I went down the same road. On the new versions of yicam, they disabled telnet and tty/uart.
> > Solution:
> > https://github.com/rage2dev/YiOutdoor
> > But we need the RT-Sdk to compile rtsp_server :/
> 
> Great contribution, thank you!
> But that means there's no actual way to flash YiHackV4 or YiHackMStar, right?
> I acquired this Yi IoT Outdoor 1080p cam thinking it was going to be easy to just have RTSP and connect it to my QNAP - seems not.

basvanrooten, 2020-06-04:
Any progress made @rage2dev ?

larsoss, 2020-07-28:
Nothing?

rage2dev, 2020-08-04:
@mitsmits: Yes, Telnet access, but we still need the RSDK.

pcrednet, 2020-08-10:
having telnet access... how could you install the firmware... if there's any for sub-camera?
At the moment I have it in the box...

Waiting for news.

Thank you for your work.

tijo82, 2020-10-20:
Is there any update on this camera?

Thanks!
Tim

basvanrooten, 2020-10-26:
@TheCrypt0 any chance this model might become supported? Really wanting to use this camera with RTSP at some point..

Any response would be greatly appreciated

paulomanuelp, 2020-12-04:
Hello, 
Any news of how to hack this version?

Abuelika, 2021-01-07:
Is there any update on this camera?

Thanks!

Kingside88, 2021-03-20:
I have the same model. It would be great if there is a chance to progess

porok, 2021-05-29:
It would be nice if you guys can help us with this camera..
If you need any sponsorship - just let me know.

cjj25, 2021-08-18:
I've created a simple RTSPServer for this model and has been confirmed working by a user :)

https://github.com/cjj25/Yi-RTS3903N-RTSPServer

Confirmed: https://github.com/cjj25/Yi-RTS3903N-RTSPServer/issues/4

ebahat, 2022-11-01:
Confirmed as working for me as well. Thanks a lot!!! 
It would be great if someone continue the effort and implement some pf the TODOs in the Git repo.