TheCrypt0/yi-hack-v4 #229: Support for Yi IoT Based Cloud Cameras (FLOUREON)

jcpuzic, 2020-05-01:
I see some chatter on an issue last year to try and support this camera that's sold on Amazon.  It doesn't see to want to read off the SD card for the firmware update.  I would like to help on the project, but just need to know what work has been done so I don't have to re-invent the wheel.  Here is a link to my camera:

https://www.amazon.com/gp/product/B07KWWW82N/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

I believe if we get this working you will see much more revenue for your RTSP server.

lordhaer, 2021-01-19:
Hi, have you managed to hack this camera?
If you could provide me with the information, I will be grateful. I can't find anything related to this camera

smoczyna, 2021-04-13:
I have similar camera from Floureon (with night vision). It is using 3rd party software you need to install on your phone and create an account to be able to use that camera. All that sucks of course and make you dependent on someone else stuff.
So I have hacked that camera instead. It basically means opening it, removing all the gats and replacing them with your own.
Camera inside my model had 30 pin FFC socket which is incompatible with everything I could use. I replaced it with Raspeberry PI camera (some gymnastics to fit it inside is required) but at the end I have my own device connected to my home network without any 3rd party extras or software. Another nice (even better) solution would be ESP32-CAM module. It is tiny and has WiFi on-board so whole piece should fit the head of the camera without any wires sticking out of it.