TheCrypt0/yi-hack-v4 #236: Embed aac audio in h264 video stream !

Vannixxo, 2020-05-22:
I’m using the camera with HomeKit and ffmpeg, and there isn’t any way to use the audio stream as it is in a separate link. I’ve got no problem with another Msstar camera using the roleo Yi hack, please add this functionality to this firmware too! The h264 should have aac stream embedded.
I also got some instability with the RTSP stream, don’t know if it could be optimized or updated some way. Thanks!

MACscr, 2020-06-30:
Ugh, im in the same boat. Cant find a solution that allows me to use both the video/audio streams to create a single viewing experience that are in sync. So frustrating

Vannixxo, 2020-06-30:
> Ugh, im in the same boat. Cant find a solution that allows me to use both the video/audio streams to create a single viewing experience that are in sync. So frustrating

Yes, please if someone knows how to do this let us know. It isn’t a matter of being in sync, I’m not really able to merge audio and video stream in the first place!

chris-chensy, 2020-07-26:
> Yes, please if someone knows how to do this let us know. It isn’t a matter of being in sync, I’m not really able to merge audio and video stream in the first place!

You can do it using video/audio mapping with the two streaming URLs. Below is my example. BUT what I can hear from the aac stream is somehow notice. I wonder if it is a problem with my camera or a bug with the code... Check if your Yi-Cam sound is OK.

`Homekit:
      [other config]...
      entity_config:
        camera.xxx:
        stream_source: -rtsp_transport tcp -i rtsp://[your IP]/ch0_1.h264 -i rtsp://[your IP]/ch0_2.aac
        audio_map: 1:a:0
        video_map: 0:v:0
        support_audio: True
`

Vannixxo, 2020-07-26:
> > > Ugh, im in the same boat. Cant find a solution that allows me to use both the video/audio streams to create a single viewing experience that are in sync. So frustrating
> > 
> > 
> > Yes, please if someone knows how to do this let us know. It isn’t a matter of being in sync, I’m not really able to merge audio and video stream in the first place!
> 
> You can do it using video/audio mapping with the two streaming URLs in the homekit setup. Below is my example via Home Assistant. BUT what I can hear from the aac stream is somehow noise. I wonder if it is a problem with my camera or a bug with the code... Check if your Yi-Cam sound is OK.
> 
> Homekit:
> [other config]...
> entity_config:
> camera.xxx:
> stream_source: -rtsp_transport tcp -i rtsp://[your IP]/ch0_1.h264 -i rtsp://[your IP]/ch0_2.aac'
> audio_map: `1:a:0`
> video_map: `0:v:0`
> support_audio: True

YOU ARE THE MAN!
Works flawlessly!!!
Thanks so much!

closed 2020-07-26 by Vannixxo