TheCrypt0/yi-hack-v4 #247: RTSP Stream does not work with homeassistant camera.play_stream

tunerooster, 2020-06-28:
I have recently installed yi-hack-v4 firmware on my Yi Outdoor Camera.  Although I am able to play the rtsp stream with, for example, ffplay, when I try to play it through HomeAssistant camera.play_media it just hangs at the "buffering" phase.  Note, that I have three other cameras (ESCAM Ant QF605) which *do* work with HA and are identically configured with the Yi camera.

CONFIGURATION
* HomeAssistant 0.111.4 running on Linux kernel 5.4.30-gentoo intel x86_64
* Yi Hack Firmware Version: 0.2.0
* Yi Hack Base Version: 3.0.0.0D_201809111054

* HA configuration.yaml:
```
ffmpeg:
  ffmpeg_bin: /usr/bin/ffmpeg
stream:
camera:
    # these are the ESCAM Ant QF605's
  - platform: ffmpeg
    input: 'rtsp://backporch/onvif1'
    name: Back Porch
  - platform: ffmpeg
    input: 'rtsp://admin:passwd@frontporch/onvif1'
    name: Front Porch
  - platform: ffmpeg
    input: 'rtsp://admin:passwd@garage/onvif1'
    name: Garage
    # this is the Yi Outdoor camera
  - platform: ffmpeg
    input: '-rtsp_transport tcp -i rtsp://frontdoor/ch0_0.h264'
    name: Front Door
```
From the HA Developer Tools -> SERVICES screen, I run:
```
Service:
camera.play_stream

Service Data (YAML, optional):
1 entity_id: camera.front_door
2 media_player: media_player.nexus9
```
When I press "Call Service" I see the following in the log:
```
2020-06-28 04:14:04 INFO (SyncWorker_12) [pychromecast.controllers] Not launching app CC1AD845 - already running
2020-06-28 04:14:04 DEBUG (SyncWorker_12) [pychromecast.socket_client] [nexus9(192.168.1.90):8009] Sending: Message urn:x-cast:com.google.cast.media from sender-0 to CastReceiver-87: {'media': {'contentId': 'https://krautclan.com/api/hls/fa3c5be4ccee813167447f54f736442f096ffb45a8afb5b59ff5c279c3217cb9/playlist.m3u8', 'streamType': 'BUFFERED', 'contentType': 'application/vnd.apple.mpegurl', 'metadata': {}}, 'type': 'LOAD', 'currentTime': 0, 'autoplay': True, 'customData': {}, 'requestId': 90, 'sessionId': '973bfc40-b873-11ea-8000-a36d1c45bc8c'}
2020-06-28 04:14:04 DEBUG (Thread-30) [pychromecast.socket_client] [nexus9(192.168.1.90):8009] Received: Message urn:x-cast:com.google.cast.media from CastReceiver-87 to *: {'requestId': 90, 'status': [{'currentTime': 0, 'media': {'contentId': 'https://krautclan.com/api/hls/fa3c5be4ccee813167447f54f736442f096ffb45a8afb5b59ff5c279c3217cb9/playlist.m3u8', 'contentType': 'application/vnd.apple.mpegurl', 'metadata': {}, 'streamType': 'BUFFERED'}, 'mediaSessionId': 47, 'playState': 'BUFFERING', 'playbackRate': 1, 'supportedMediaCommands': 15, 'volume': {'level': 1, 'muted': False}}], 'type': 'MEDIA_STATUS'}
2020-06-28 04:14:04 DEBUG (Thread-30) [pychromecast.controllers] Media:Received status {'requestId': 90, 'status': [{'currentTime': 0, 'media': {'contentId': 'https://krautclan.com/api/hls/fa3c5be4ccee813167447f54f736442f096ffb45a8afb5b59ff5c279c3217cb9/playlist.m3u8', 'contentType': 'application/vnd.apple.mpegurl', 'metadata': {}, 'streamType': 'BUFFERED'}, 'mediaSessionId': 47, 'playState': 'BUFFERING', 'playbackRate': 1, 'supportedMediaCommands': 15, 'volume': {'level': 1, 'muted': False}}], 'type': 'MEDIA_STATUS'}
```
When I re-run the same configuration, but with camera.garage (the ESCAM rather than the YI) I get the same log output, with an additional line as follows:
```
 2020-06-28 04:12:24 DEBUG (Thread-30) [pychromecast.socket_client] [nexus9(192.168.1.90):8009] Received: Message urn:x-cast:com.google.cast.media from CastReceiver-87 to *: {'requestId': 0, 'status': [{'currentTime': 0, 'mediaSessionId': 46, 'playState': 'PLAYING', 'playbackRate': 1, 'supportedMediaCommands': 15, 'volume': {'level': 1, 'muted': False}}], 'type': 'MEDIA_STATUS'}
```
So it appears the YI cam RTSP stream is causing the media_player to hang on Play State: *BUFFERING*.

I have tested with several media_player's including a *Shield Android TV* and a *Google Home Hub* display, all with the same results...  they hang with a spinning circle saying "Video loading...".
They also all work with the ESCAM RTSP streams.

​Please advise...

plutec, 2020-07-05:
Hi,
I have the same problem with a Yi Outdoor. The RTSP apparently does not work fine. Some times, after hard reboot (cable off) the RTSP server does not start (nmap -p 554 IP -> Close). I need, in configuration, remote the RSTP support, reboot, enable the RSTP and reboot again to gain access to the service.

chilman408, 2020-09-04:
hi guys, 

off topic but I can't find this information. 

how much do you have to donate to get rtsp server?

also is it on a per camera basis?