TheCrypt0/yi-hack-v4 #248: It is impossible to flash the Yi Home 1080p Camera

Dragon0501, 2020-07-01:
I copied the files to the root directory of the SD card. After this I inserted the SD and cut the AC. After reconnecting it starts normal the original firmware. No flashing or other... Pleas help me!

jflemieux, 2020-07-02:
If your firmware version is 8.2.0, use this version : https://github.com/roleoroleo/yi-hack-Allwinner

Dragon0501, 2020-07-02:
Which version do I have to use? 
[image]

jflemieux, 2020-07-02:
Your version is the same I had before flashing. Follow the instructions on the repo I previously linked to ( I used the y20gam_0.1.6.tgz file )

Dragon0501, 2020-07-02:
Will give it a try! Thank you!

Dragon0501, 2020-07-02:
thank you very much! worked without a problem!!!

closed 2020-07-02 by Dragon0501